Ultimaker/Cura #3823: On one machine missing file menu in GUI. Because of that there is a difference between button's picture area and button's active area(to click at button i have to move cursor for some pixels lower the button)

Unouzhukrnet, 2018-05-17:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2018-05-17:
Probably a duplicate of https://github.com/ultimaker/cura/issues/2068 (in short: you have to update your intel GPU driver), but I can't be sure because you ignored the issue template and the issue title is not a good place to explain what you mean.

Unouzhukrnet, 2018-05-17:
On one machine missing file menu in GUI. Because of that there is a difference between button's picture area and button's active area(to click at button i have to move cursor for some pixels lower the button)
[image]
I tried to reinstall and also to clear appdata folder but result is the same.

closed 2018-05-17 by Unouzhukrnet