Ultimaker/Cura #3826: Source_User field in Cura Connect

jkennard, 2018-05-17:
In previous versions of Cura (<3.0) we could see the user name for the print job via the API page for each printer. It seems that with Cura Connect that field is now "hidden" in the sense that this field now will only display the name of the hub printer, instead of the name of the user who submitted the job. Is there any way to still get at this information via the API?

**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Going to http://10.180.1.209/api/v1/print_job shows the results below. (IP address here is an example)

**Actual Results**
Source_user field shows the name of the printer/printer hub of Cura Connect

**Expected results**
Source_user field should show the name of the user who submitted the print job

ChrisTerBeke, 2018-05-18:
Do you have the issue in the Cura UI or the Cura Connect web interface?

ChrisTerBeke, 2018-05-19:
Cura Connect is not really supported on GitHub, so all I can do is relay this information to the developers on that team.

closed 2018-05-23 by ChrisTerBeke

Ghostkeeper, 2018-07-02:
You should still be able to get the user from `<ip>/connect-api/v1/print_jobs`.

jkennard, 2018-07-02:
Actually that does not. It only takes you to a page (<ip>/default.html) where you can click on Cura Connect, Event Log, API Documentation, etc. If you go to <ip>/api/v1/print_job/source user this will only display the machine name of the hub printer for Cura Connect.

Prior to using Cura Connect we could easily pull the user name from any of these API urls but now it seems to be hidden, or there is some other place that we need to pull this information from. The Cura Connect page displays the user name correctly so why can I not get to it in the same way (via API) as we have done in the past? I have read through the API documentation but there doesn't seem to be any information that relates to changes between pre-v3. 0 and Cura Connect. Any ideas?

Jerrod

Disclaimer

The information contained in this communication from the sender is confidential. It is intended solely for use by the recipient and others authorized to receive it. If you are not the recipient, you are hereby notified that any disclosure, copying, distribution or taking action in relation of the contents of this information is strictly prohibited and may be unlawful.

This email has been scanned for viruses and malware, and may have been automatically archived by Mimecast Ltd, an innovator in Software as a Service (SaaS) for business. Providing a safer and more useful place for your human generated data. Specializing in; Security, archiving and compliance. To find out more visit the Mimecast website.

Ghostkeeper, 2018-08-13:
Indeed this source user will now display the name of the host printer, by the looks of it. That's not very useful.

I am able to get the print job owner using this URL though: `<ip>/cluster-api/v1/print_jobs/`. This lists some history though (up to a certain limit determined by the file size of your g-codes). To get the currently running prints, you'd need to filter them by status: "printing" and to zone in on a specific printer you'd need to know that printer's UUID.