Ultimaker/Cura #3828: [3.3.1] Adaptive layers causes gaps when slicing

waechtjn, 2018-05-17:
**Application Version**
3.3.1

**Platform**
Windows 10.0.16299 Build 16299 (2018-05-10)

**Printer**
Crealty CR-10 
[Project file and STL model](https://github.com/Ultimaker/Cura/files/2014334/Files.zip)

**Steps to Reproduce**
- Add the model from the STL file in the attached ZIP archieve
- Rotate the model by 90° so that the round/dome like part faces upwards:
[image]
- Slice using 0.2 layer height and adaptive layers with maximum variation of 0.1

**Actual Results**
The sliced part contains a gap:
[image]

**Expected results**
The sliced part should not contain gaps (picture is for a maximum variation 0.099):
[image]

**Additional Information**
There are no gaps when slicing without adaptive layers or when setting maximum variation to 0.099 (as in the picture above).

The (ugly) gaps are actually visible in the printed result as well.

ChrisTerBeke, 2018-05-18:
There's some known issues with Adaptive Layers. We won't be working on this soon though.

Ghostkeeper, 2018-06-21:
This is not a problem with adaptive layers, as far as I can see, but only shows up with adaptive layers because it is slicing at a different Z coordinate where this happens to occur.

The real problem is that the polygon is like 2.25 line widths thick in that place, so it manages to fit 2 lines there but not the third.

waechtjn, 2018-06-21:
I agree: the problem is also present without using adaptive layer height.

You can reproduce it using the same model as above by setting layer height to 0.225. This time the gap is at the top (and, thus, less visible on the screenshot; however, you can see the infill):
[image]

Ghostkeeper, 2018-08-06:
Looking at this open issue again, I suppose this sort of thing should get filled as tiny gaps, shouldn't it?

If there is a small gap of less than half a line width, I don't really see what a slicer could do to fill that short of performing a physics relaxation to spread out the existing lines better, which is infeasible for the allowed computation time.

Is Tiny Gaps enabled in this print?

GregValiant, 2026-09-06:
I'm cleaning up these old reports.  I'll go ahead and close this as the Arachne Engine in current versions cures this problem.

closed 2026-09-06 by GregValiant