Ultimaker/Cura #3829: Unexpected "G0 X0 Y0" during layer changes?

AbeFM, 2018-05-17:
I keep finding my printer "homing" during prints.

**Application Version**
3.3.1

**Platform**
win10

**Printer**
Custom Dual Ext/Marlin

**Steps to Reproduce**
Slice!

**Actual Results**
`G0 F7200 X184.995 Y150.007
G1 F2700 E6.60243
G92 E0
G1 X110 Y130
T1
G92 E0
M109 S190
M104 T0 S210
M106 S76.5
G1 F2100 E1.4
G0 F7200 X183.357 Y150.242 Z8.62
G0 X174.983 Y151.563
G0 X154.922 Y153.664
G0 X154.271 Y153.618
G0 X153.798 Y153.011
G0 X154.144 Y151.054
G0 X154.427 Y150.947
M104 S192
;TYPE:SUPPORT
G1 F2100 E4
G1 X154.301 Y150.89 E4.00598
G1 X154.174 Y150.725 E4.01498
G1 X152.793 Y151.228 E4.07853
G1 X154.249 Y151.957 E4.14894
G1 X154.427 Y150.947 E4.19328
G1 F2100 E1.59328
G0 F7200 X154.108 Y151.025
G0 X152.242 Y151.707
G0 X151.631 Y151.383
G0 X0.00 Y0.00  **************************************************
;TIME_ELAPSED:328.972004
;LAYER:33
M104 S188
G0 X0.00 Y0.00 Z8.88  **************************************************
G1 F2100 E0.19328
G92 E0
G1 X110 Y130
T0
G92 E0
M109 S210
M104 T1 S190
M106 S153
G1 F2700 E7.4
G0 F7200 X0.00 Y0.00 Z8.88
G0 X114.514 Y149.174
G0 X114.951 Y149.94
G0 X115.001 Y150.012
G0 X115.094 Y149.971
M104 S205
;TYPE:WALL-OUTER
G1 F2700 E10`
Marked offending rows with   **************************************************
[CFDMP_AbeStringTest1.curaproject.zip](https://github.com/Ultimaker/Cura/files/2014701/CFDMP_AbeStringTest1.curaproject.zip)

**Expected results**
NOT going to 0,0?

ianpaschal, 2018-05-18:
I can't find the issue back but this is a duplicate and we actually discussed it yesterday and decided we were not going to devote time to fixing this. Sorry about that.

ianpaschal, 2018-05-18:
At least I thought that's what we agreed, but it's still in our sprint.

Devs, see: CURA-5179

smartavionics, 2018-05-18:
Possibly fixed in https://github.com/Ultimaker/CuraEngine/pull/776 ?

closed 2023-01-06 by nallath