Ultimaker/Cura #3830: Scaling parameters bug

kevinjones4642, 2018-05-17:
Cura 3.3.1

Win7 64

Ultimaker 3

1). Load a part in Cura.
2). Rotate the part. For example, around the x-axis by 90 degrees.
3). Scale the part in Z with the Uniform Scaling box un-checked.

This seems to scale the part correctly, but it reports in the scaling widget that both y and z have been scaled by 102%, but the actual length of the part in y appears to be unchanged.

Cura should report that x was scaled by 100%, y was scaled by 100% and z was scaled by 102%

It appears as if the scaling percentages reflect the un-rotated part, but the dimensions reflect the rotated part

ianpaschal, 2018-05-18:
This is in progress. Devs, see: CURA-5307

Ghostkeeper, 2018-06-21:
Yeah this is a duplicate of https://github.com/Ultimaker/Cura/issues/3625.

closed 2018-06-21 by Ghostkeeper