Ultimaker/Cura #3833: Show Configuration Folder - Not Working

goks95, 2018-05-18:
I have installed the 3.2.1 and the latest one as well. And I wanted to load my 3dprinter json file into the configuration folder and the "show the configuration folder" under "Help" tab doesnt seem to be working. 
How can I fix this? Thanks

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

goks95, 2018-05-18:
Yeah.

ChrisTerBeke, 2018-05-18:
Mm this is hard to debug without the logs, which are in the configuration/data folder. You can find details on finding the folder manually here: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations. Please share the logs in that folder.

goks95, 2018-05-18:
Got it fixed. Thanks!

closed 2018-05-18 by goks95