Ultimaker/Cura #3834: [3.3.x] Feature request:  per-filament and per print profile g-code

VanessaE, 2018-05-18:
Use cases:

For filament g-code, one could control various firmware features that differ on a per-material basis, such as linear advance or firmware retraction (which still depends on an external plugin that doesn't work in 3.3.1 for Linux :frowning_face: ).

For print profiles (e.g. speed/quality/etc), to be able to override the above when needed (for example, on my bot, I have to set linear advance to 0 when printing at high speeds).

Better to have the slicer insert the appropriate commands than to have to remember what to send when.

ChrisTerBeke, 2018-05-18:
I don't follow... It's already possible to change settings based on a material or quality profile, this includes speed and retractions. What settings/commands would need to be added exactly to meet your demands? Maybe describe with some G-code snippets?

VanessaE, 2018-05-18:
Well, on my bot for example, I would want to use `M900 K0.15` to set an appropriate Marlin linear advance value for Atomic white PLA, when printing at moderate speeds, but say `M900 K0.35` or thereabouts for PETG.  I imagine TPU or other flexibles would need even higher values (though I don't use them, yet).  I have the K0.15 setting in my printer start G-code already, just to at least compensate for the extruder and hotend (but fixed at that value, it can't compensate for differences in filaments).

ChrisTerBeke, 2018-05-22:
So it boils down to having a start/end g-code that's specific for the profile?

VanessaE, 2018-05-22:
Yep, that's about the size of it.

ChrisTerBeke, 2018-05-22:
Alright, thanks for the clarifications.

fieldOfView, 2018-05-22:
This is already possible by editing the config files, there is just no way to edit gcode snippets in the sidebar. An important reason why this is not currently possible in the frontend is that there is no multiline string setting widget for the sidebar.

If you change the start gcode in the Machine Settings, it will be stored in a .inst.cfg file in the definition_changes folder inside the configuration folder. If you want to move this start gcode to a quality profile, find its file inside the quality_changes folder, and move the machine_start_gcode lines from the definition_changes .inst.cfg to the quality_changes file.

genghisnico13, 2018-05-23:
I would like this feature too. Right now I have a section for each filament I use in the machine start g-code and uncomment the one I'm currently using. I set the K factor for linear advance and the extruder steps/mm (how deep is the filament being bitten).

filipgoc, 2018-05-24:
@fieldOfView That was the single one most useful answer one could give here. Thumbs up.

jackha, 2018-06-20:
We've discussed it and decided not to do do this. However, @VanessaE you can always add multiple printers with each of them with their own start/end g-code.

VanessaE, 2018-06-20:
The whole point of material start/end g-code is to be able to send material-specific commands without having to remember what goes with what.  That's the *computer's* job.

Most people I know who have a 3d printer have more than one - typically with different styles and capabilities (think high-end delta vs. CoreXY vs. a cheap Prusa i3 MK1 derivative.   How's that going to work out?

fieldOfView, 2018-06-21:
@Ghostkeeper, there's a nifty plugin here: https://github.com/fieldOfView/LinearAdvanceSettingPlugin
It adds a couple of settings to the sidebar that can be used in start/end gcode snippets.

Ghostkeeper, 2018-06-29:
Nice, I didn't know about that one.

It looks like that implements the use case that was the underlying issue here then. For the issue itself, we would prefer to keep the start/end g-codes a machine setting rather than a normal setting.

closed 2018-06-29 by Ghostkeeper