Ultimaker/Cura #3835: [3.3.1] Project Summary Box Problem

fatalelieberi, 2018-05-19:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Tevo Tornado - Custom FDM
[MyPrinter.curaproject.zip](https://github.com/Ultimaker/Cura/files/2019056/MyPrinter.curaproject.zip)

**Steps to Reproduce**
1. Open menu File -> Save Project

**Actual Results**
Unclear writings inside the Project Summary
[image]

ianpaschal, 2018-05-22:
Indeed. No doubt this is caused by the dialog having a fixed size dialog which can't fit 8 extruders. Simple fix is putting the summary in a small scroll area.

ianpaschal, 2018-05-22:
Internal ticket created, devs see CURA-5385.

ianpaschal, 2018-05-28:
We've fixed this; will be in 3.4 released in June.

closed 2018-05-28 by ianpaschal