Ultimaker/Cura #3836: [3.3.1] Cura Lithophane not working as intended.

modem7, 2018-05-19:
When loading a picture into Cura, it seems to slice until a certain point, then stick to 100% CPU usage with no further movement. 

[image]

I've attempted to wait for around 20 minutes with no movement on slicing and curaengine @ 100% CPU.

ChrisTerBeke, 2018-05-19:
logs please (as requested in the issue template).

modem7, 2018-05-19:
[stderr.log](https://github.com/Ultimaker/Cura/files/2019826/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/2019827/stdout.log)
[3.3.zip](https://github.com/Ultimaker/Cura/files/2019828/3.3.zip)

jackha, 2018-07-16:
No activity and not reproducible, closing.

closed 2018-07-16 by jackha