Ultimaker/Cura #3838: [3.3.1] Target specific areas of model for adaptive layers

mechaMecha, 2018-05-20:
I've been really impressed with Cura's development over the past year or so. It has become my default slicer because it gives me so much control over just about every print setting, and its GUI is so much more intuitive than its competitors.
 
Adaptive layers has been a great new feature, but I've found that its usefulness is limited to fairly symmetrical models without a lot of detail, or if detailed areas are isolated to their own vertical spaces within a model. Adaptive layers would be so much more useful if there could be a way to target specific areas on a model for Cura to include in calculating the step-down of layer thickness. (Or, conversely, to target specific areas for Cura to exclude when it calculates that.)
 
The way that the new "support blocker" tool works seems pretty nice; maybe "adaptive layer include/exclude" could work in a similar manner?

ianpaschal, 2018-05-22:
This is a `Won't Do`. Changing the adaptive layer heights in certain spots like the support blocker completely breaks the idea of "layers" and the FDM workflow.

closed 2018-05-22 by ianpaschal

mechaMecha, 2018-05-24:
I'm not sure why this would be considered a "won't do." Maybe I'm not explaining myself properly. I'm not talking about "changing" adaptive layer heights in certain spots, I'm saying adaptive layers would be much more useful if we could tell Cura to simply ignore certain details in the model.

[image]

For example, if I were much more concerned about eliminating the visible banding on the exterior of this very simple object than having the arched opening printed as nicely as possible, it certainly would be nice to tell Cura to ignore the arched opening when calculating adaptive layers. It would reduce the print time as well. Obviously the problem is exacerbated on more complex models.

Mentioning the manner in which support blocker has been implemented was simply a very quick idea of how it might work.

ianpaschal, 2018-05-24:
No, I get it, but then what do you get where that feature was? A hole in the geometry? New geometry? What if it's not watertight? This would require a massive re-write of the slicer.

mechaMecha, 2018-05-24:
OK. I guess I'm just misinterpreting the "Won't Do" label that my suggestion was tagged with. So this is more of a "can't do," I guess? I apologize, I'm not a regular GitHub user, just a heavy-duty Cura user.

Even if it could be implemented in sort of a "change at z" manner, where a portion of the model receives fixed layers and another receives adaptive layers, it would make the feature considerably more useful, I think.

mechaMecha, 2018-05-24:
OK. Thank you for taking the time to reply!