Ultimaker/Cura #3839: New Bridfig feature causes briddle layers

b14ckyy, 2018-05-20:
**Application Version**
Cura 3.3.1

**Platform**
Windows 10 Pro x64

**Printer**
Anet A8 - Modified with e3D v6 Hotend, Bowden Extruder,  

**Steps to Reproduce**
Print an object with thin pillars and bridges between them. Added an STL Model I designed where it is easy to reproduce, also the gcode and cura Project file.

[CFDMP_Gimbal Camera Mount v9.curaproject.zip](https://github.com/Ultimaker/Cura/files/2020678/CFDMP_Gimbal.Camera.Mount.v9.curaproject.zip)
[CFDMP_Gimbal Camera Mount v9.zip](https://github.com/Ultimaker/Cura/files/2020679/CFDMP_Gimbal.Camera.Mount.v9.zip)
[Gimbal Camera Mount v9.zip](https://github.com/Ultimaker/Cura/files/2020680/Gimbal.Camera.Mount.v9.zip)

**Actual Results**
the layer, where the bridging happens seems to have full reduced Extrusion to 50%. So the outer wall has one layer with many small holes in it. This causes to have weak Points in layers. 

**Expected results**
Normal layer Extrusion before and after the Bridge.

**Additional Information**
Picture of print:

[image]

jackha, 2018-06-19:
@b14ckyy:
- Does the lag of filament flow start about 3 mm before the bridge? Or is it still the whole layer?

- What results do you get of you turn bridging off? I'm mostly wondering if you still have the under extrusion.

smartavionics, 2018-06-19:
My experiments have shown that step changes in extruder rate such as are used when the bridge segment is printed with a different flow or speed to the normal wall have a big impact on wall visual quality. So if you want to avoid these underextrusions, you need to not change the speed or flow for the bridge wall segments compared to the normal settings. If your printer/filament/settings combination cannot print the bridge walls successfully using the normal speed/flow then you have the problem to decide whether you want to preserve the visual quality of the non-bridge segments of the wall line or you want the bridge segments of the wall line to be better. It's a tradeoff.

jackha, 2018-06-19:
Let's see what @b14ckyy experiences the same. Then we may be able to close this issue.

b14ckyy, 2018-06-21:
Hi. My Printer was not running since a few days. I'll search a nice test model to try different Settings. I'll report back asap

Ghostkeeper, 2018-07-31:
Any luck?

diegopradogesto, 2019-09-11:
Long time waiting for a response. Feel free to reopen if this still reproduces for you.

closed 2019-09-11 by diegopradogesto