Ultimaker/Cura #3840: Combing doesn't use retraction above skin when it should

smartavionics, 2018-05-21:
**Application Version**

Master branch (+ I think 3.3.1 also)

**Platform**

Linux

**Printer**

Any, this example using S5

**Steps to Reproduce**

Load appropriate model, set combing to "no skin", slice.

**Actual Results**

Some travel paths on the first few layers are not using retraction when they should.

**Expected results**

All short travel paths that cross skin regions should use retraction.

**Additional Information**

Here's the first layer (with skin regions), note how some of the travel paths do not use retraction.

[image]

Here's the same slice looking at the first layer that doesn't have any skin, note how all the travel moves between parts are using retraction.

[image]

Here's the project.

[UMS5_fish_fossilz.curaproject.zip](https://github.com/Ultimaker/Cura/files/2022724/UMS5_fish_fossilz.curaproject.zip)

smartavionics, 2018-05-23:
That shows great faith, the PR hasn't been "approved" yet!

ChrisTerBeke, 2018-05-23:
Haha, just cleaning up a bit. PR is also an ticket in our internal system.