Ultimaker/Cura #3841: [Cura 3.3.1] Start Gcode issue

modem7, 2018-05-21:
Cura 3.3.1

My current gcode is as follows:

M107                                ;start with the fan off
M140 S{print_bed_temperature}       ;Start heating bed
M190 S{print_bed_temperature}       ;Wait for bed to reach temp before proceeding
M104 S{print_temperature}           ;Start heating extruder
G28                                 ;Home all
G29 A                               ;Run auto levelling
G1 Z5 F1800                         ;lift Z by 5mm
M109 S{print_temperature}           ;Wait for extruder to reach temp before proceeding
G92 E0                              ;zero the extruded length
G1 F200 E3                          ;extrude 3mm of feed stock
G92 E0                              ;zero the extruded length again

However, what I'm seeing in the gcode is:

;Generated with Cura_SteamEngine 3.3.1
M190 S60
M104 S215
M109 S215
M82 ;absolute extrusion mode
M107                                ;start with the fan off
M140 S60       ;Start heating bed
M190 S60       ;Wait for bed to reach temp before proceeding
M104 S215           ;Start heating extruder
G28                                 ;Home all
G29 A                               ;Run auto levelling
G1 Z5 F1800                         ;lift Z by 5mm
M109 S215           ;Wait for extruder to reach temp before proceeding
G92 E0                              ;zero the extruded length
G1 F200 E3                          ;extrude 3mm of feed stock
G92 E0                              ;zero the extruded length again
G92 E0
G1 F6000 E-0.5

It seems that it's entering the m190, 104 and 109 codes in prior to the gcode. 

Am I being a muppet, is this a bug, or an "intended feature"?

Thanks!

[Relevant files.zip](https://github.com/Ultimaker/Cura/files/2023306/Relevant.files.zip)

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2018-05-21:
Instead of ```M140 S{print_bed_temperature} ;Start heating bed
M190 S{print_bed_temperature} ;Wait for bed to reach temp before proceeding
M104 S{print_temperature} ;Start heating extruder```, use ```M140 S{material_bed_temperature} ;Start heating bed
M190 S{material_bed_temperature} ;Wait for bed to reach temp before proceeding
M104 S{material_print_temperature} ;Start heating extruder```

modem7, 2018-05-21:
But this would be the material temperature set in the material profile rather than the temperature I set in the GUI is that correct? Or am I reading it incorrectly?

fieldOfView, 2018-05-21:
No, this is not correct. The value you see in the sidebar always overrides the value set in the material. Also, the deprecated print_temperature references the same value as material_print_temperature internally (but it does not make Cura believe you have added your own preheat sequence).

fieldOfView, 2018-05-21:
Sorry, I missed the M109 later down the gcode, also replace the {pattern} over there for consistency.

modem7, 2018-05-21:
Ahhah! Appreciate it!

Will give it a test ASAP, thank you

modem7, 2018-05-21:
Perfect, all seems to be working. 

Appreciate the help!

closed 2018-05-21 by modem7