Ultimaker/Cura #3843: [3.3.1] Coast not being applied to infill of second part

VanessaE, 2018-05-22:
Sounds simple, right? ;-)

In the screenshot below, if you look at the X-shaped infill in the cylinder (which I refer to as a "dwell tower" because its purpose is to waste time) under the Axes indicator, you can see that it's all there - the infill of this part is not being affected by coasting settings (highlighted in yellow).  Yes, I know the coasting amount is pretty extreme, but this is something of an experiment anyway (I was originally working with a worn-out hotend, but it's since been replaced and my original reason for coasting went away).

3mf (zip) of the plate:  [PI3_excavator_extended with mouse ears and dwell tower.3mf.zip](https://github.com/Ultimaker/CuraEngine/files/1993117/PI3_excavator_extended.with.mouse.ears.and.dwell.tower.3mf.zip)
Cfg file: [cura.cfg-20180510.zip](https://github.com/Ultimaker/CuraEngine/files/1993124/cura.cfg-20180510.zip)  (note: while this .cfg was where the instructions in this issue tracker said it would be, I don't see anything that looks like a log file)

**Application Version**

3.3.1

**Platform**
 
Debian 9.3, AMD64

**Display Driver**

Standard issue Gallium driver (AMD R9 280x GPU driving 4 displays)

**Steps to Reproduce**

Load the attached .3mf.
Make sure the dwell tower has enough infill to see the coast effect clearly.
Enable coasting, and set a fairly extreme amount similar to my screenshot
Slice it.

**Actual Results**

The infill of the dwell tower is not affected by the coasting setting.  For that matter, some of the inner (green) perimeters here and there aren't affected, either, even though the minimum volume is definitely set small enough.  My workaround for this particular project was to just set that dwell tower to be hollow and adjust the perimeter speeds to achieve my aims.

**Expected results**

Coast should be applied to all line types?

**Additional Information**
[image]

(this issue copied over from CuraEngine because it got no attention there.  Maybe I misfiled it?)

VanessaE, 2018-06-17:
Come to think of it, this isn't limited to infill, or to a second part.  Affects perimeters too, even if it's just one part, at least in 3.4.0 beta:

[image]

(note the islands at the far lef, and below the left-most circle)

The object is the jet fighter model from https://www.thingiverse.com/thing:2887652 (the original 100% size one)

Ghostkeeper, 2018-06-28:
Coasting was intended to work only on the walls as a way to make the seam nicer by depleting the nozzle pressure before finishing up the wall. It will not work on infill, skin, brim, support, etc.

That second post is a bug. I can reproduce it as well.

Ghostkeeper, 2018-06-29:
We decided to not give this issue any priority right now because coasting is still experimental. It'll get picked up when we do another pass to make the feature more stable in general.

gieljanv, 2020-05-28:
Hi @Ghostkeeper , Does Coasting still not work on infill lines?

Ghostkeeper, 2020-05-29:
Indeed, it still doesn't get applied to infill lines, only to walls.

thanhtacles, 2020-06-12:
Bump this issue back up, coasting is a household name in good slicing practices and should be recognized as such. It actually does more than seams, it helps reduce stringing too. I'm getting some stringing on small thin parts and I think coasting ever so slightly will help with this.

Corallus-Caninus, 2021-08-18:
For infill lines consider the wiping setting it is the equivalent of coasting.

Ergonomicmike, 2023-07-24:
I would like to revive this Issue and raise it to a Feature Request: That Cura allow the option of "Replace Retraction with Combing."  ([I'm not the first to request this](https://community.ultimaker.com/topic/32971-coasting-in-cura-on-layer-change/), BTW.)

I respectfully submit that Coasting is better than Retraction.  And so Cura should be the first to take full advantage of Coasting.

As most all tutorials mention, Retraction has a "fatal flaw" (my words). It does not pull a suction on the filament in the nozzle. So like when I "flow,"  there's always some dribble after I command the flow to stop. 

Since there will always be dribble after the fact with Retraction, the other flaw (IMO) is that Retraction always starts too late. It doesn't start soon enough to end the dribble on time. If nothing else, Retraction should start slightly ahead of when extrusion stops. (Like Coasting already does.)

Using Retraction to minimize dribble hasn't been much of an issue in the past. But it is more of an "issue" nowadays with the advent of high-flow nozzles (less viscosity of the plastic) and faster print speeds (increased inertia/momentum due to higher mass flows). So the dribble is more pronounced now.

I noticed this after upgrading my lowly Ender 3 v2 with a 0.4 mm Copper nozzle. The temperature at the tip of the nozzle is now hotter than it was with my standard brass nozzle. So now the plastic at the tip is less viscous than it was. So even when printing at slow speed (where the momentum of the flowing plastic hasn't increased) more plastic dribbles after Retraction due to better heating of the nozzle. Increasing Retraction doesn't solve this problem.

(There is also more dribble after Retraction with my 0.6 mm CHT high-flow nozzle although that's because retraction doesn't work well on a CHT, as even Bondtech acknowledges. But I suspect that even non-high-flow 0.6 mm nozzles ooze more since adhesion to/capillary action in the nozzle opening is probably less than it is in small diameter nozzles. Still, if Cura were to offer Coasting instead of Retraction, CHT users would be served too.)

All these effects are most obvious when one sets "Retract at layer change."

I noticed this when I needed a part without a seam. Spiralize didn't give me the bond I needed with the flange. And Coasting is not yet implemented for layer changes. So the print head comes to a stop in the horizontal, then a retraction - with its inevitable dribble - and then the jump in Z. (Does the Retract step cause a "double stop"?) 

This invariably leaves a seam, albeit a small one, no matter how crazy high the Retraction.

It also causes stringing during travel. 

[image]

I didn't have this before the change to a high flow Cu nozzle.

Whereas Coasting, when tuned correctly, leaves no seam because Coasting puts the dribble to good use. It basically uses up the ooze. 

As a dev said above, "Coasting was intended ... as a way to make the seam nicer by depleting the nozzle pressure before finishing up the wall." And it now does this perfectly.

But Coasting is only triggered by travel. This means that it isn't triggered during layer changes, when I would like it. 

[image]

Therefore, my request that Cura allow users to "Replace Retraction with Coasting."

From the perspective of "Nothing is impossible for the man who doesn't have to do it himself," I presume that the existing Coasting code is mature now and that "all" that would have to be done is to call the Coasting code whenever Retraction is currently called.

Although there is one corner case (that I can think of) where Coasting can't replace Retraction. And that's where there's just one "dot" of extrusion, say at the point of a nose cone of a rocket. For that special case, Retraction is the only option, since there's no previous flow to coast from. So there would have to be a check in the code for that corner case.

I don't see a downside to allowing users to "Replace Retraction with Coasting." (Except for the extra work for the Cura team. Which I am not dismissing.)

Alternatively, if the Cura team still doesn't want to offer this option, perhaps a Contributor reading this could write a post-processing plug-in that would parse the g-code to find Retractions. After checking for the corner case above, it would cut and paste the Retraction command to move it to back n-1, or n-2 (user set-able) in the g-code so as to start Retraction in advance. Sort of a hybrid between Coasting and Retraction. (Call it "Coast-Raction"?)

If I were younger and if I knew how to code, I would do it. But, as you see, I am already dribbling.

Ergonomicmike, 2023-07-28:
Also, replacing Retraction with Coasting would also speed up prints. 2 hours saved (out of 12 hours), as I contemplate the Torture Toaster print.