Ultimaker/Cura #3847: *.3mf's may load in with dead volume

AbeFM, 2018-05-22:
Repost from: https://github.com/Ultimaker/CuraEngine/issues/778
Reposted from https://community.ultimaker.com/topic/22569-bed-not-detected-right-in-slicingpositioning/

**Application Version**
2.? - 3.3.1

**Platform**
win10

**Display Driver**
NVidia 388.73

**Steps to Reproduce**
Load a *.3mf file. (Might happen on solidworks import, too, unsure)

**Actual Results**
SOME models, (for example: [DHookP110_110_120.zip] (https://github.com/Ultimaker/CuraEngine/files/2019602/DHookP110_110_120.zip) ) when loaded seem to refrence an invisible "origin" and CURA treats the object as if it were a solid extended body. In fact, in solid body mode, the limits are shown, and in many cases I not only cannot get the part onto the bed without manually positioning to a few microns, but look at how it is treated:
[image]
As the object, and it's dead volume, won't fit on the bed, I can't even slice this hook.

**Expected results**
Object's boundaries are determined from triangles in body. Save item as .stl and it opens and behaves normally

**Additional Information**
I've made complaints on the forums in the past about this behavior, but only noticed it was with .3mf files recently.

ChrisTerBeke, 2018-05-23:
It seems the 3MF file has been 'grouped' and exported that way. Some programs (like MS 3D Builder) do that. You can right-click on the part to ungroup it and then auto center the object. In short: the model was exported by the design software incorrectly.

ChrisTerBeke, 2018-05-27:
If a 3MF is constructed in this particular way, Cura has to read it like this, as 3MF is a standard so we can't just make up behaviour for this. It's more an issue with the modelling software is a 3MF was exported like this in the first place.

Ghostkeeper, 2018-06-28:
We had a bug in Cura at some point that the coordinate origin (0, 0, 0) was always included in the bounding box in 3MF. Maybe that is back. The 3MF file still loads wrong in current Cura.

In other words, it is correct that Cura loads the 3MF file as a group. It is not correct that the bounding box is not close to the model.

AbeFM, 2018-06-28:
THAT sounds right to me. It reflects what I'm seeing.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2018-08-09:
This is fixed for version 3.5!

closed 2018-08-09 by Ghostkeeper