Ultimaker/Cura #3851: has_custom_materials does not hide materials by default anymore?

filipgoc, 2018-05-25:
Hi, 

It seems that Cura now includes all materials that do not exclude the printer (hardware compatible) by default.*

It used to be that when using has_custom_materials, Cura would show _only_ the materials that I explicitly added the printer to. 

**Is this a bug or a new feature?** How is the system designed to work now? 

Thanks!

filipgoc, 2018-05-25:
PS: I can see that there's an option to 'exclude_materials' in the definition file, but this does not hide materials added in the future, and it would be a lot of work to exclude everything (in addition to generics)

fieldOfView, 2018-05-25:
Duplicate of #3603, which has been marked as a bug but has not yet seen progress.

closed 2018-05-25 by ChrisTerBeke

filipgoc, 2018-05-25:
Right! Sorry for duplication, I missed that bug report.