Ultimaker/Cura #3852: [3.3.1] Unexpected temperature change

Renha, 2018-05-25:
**Application Version**

Cura 3.3.1
**Platform**

Windows 10
**Printer**

Custom Delta
**Steps to Reproduce**

just slice attached project
**Actual Results**

printer pauses sometimes mid-print and waiting, ruining surface finish with nozzle heat
**Expected results**

printer prints without pausing
**Additional Information**

There are lines like:
```
M104 S241
M109 S246
M104 T0 S246
M104 T1 S246
```
in GCODE, while printing and standby temperature settings are all set to 256 for both extruders.
[renha.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2039259/renha.curaproject.3mf.zip)

Renha, 2018-05-29:
as a workaround i've added post-processing scripts changing S241 and S246 to S256, it worked for me.

But problem itself still exist, what if on another model these temperatures would be different?

diegopradogesto, 2018-05-29:
There are two other settings you may want to take a look at: _Initial Printing Temperature_ and _Final Printing Temperature_. As you can see in your project file, the settings are set to 246 and 241 so you can tweak them a bit to get good results:
[image]

Renha, 2018-05-30:
thanks, didn't think about hidden settings

closed 2018-05-30 by Renha