Ultimaker/Cura #3853: [bug?] Gradual infill settings in Recommended tab do not respect default profiles.

filipgoc, 2018-05-25:
**Application Version**

3.3.1

**Platform**

Mac

**Printer**

JellyBOX, other custom...

**Steps to Reproduce**

Use a profile that has gradual infill defined as a default. In my profile for example, I'm using
```
gradual_infill_step_height = 1
gradual_infill_steps = 1
infill_pattern = cubic
infill_sparse_density = 70
```

👍 In the recommended tab, this is what is shown when there are no overrides in place: 
[image]

👍 Then, deactivate gradual infill. So far so good. 
[image]

👎 Now, activate the infill again. The settings do not go back to the last used or what's in the profile, but rather somewhere completely else. No clue where. 
[image]

Funny thing is, Cura even knows this is an override as shown by the round arrows. 
[image]
[image]

**Expected results**

I'd  expect the settings to just switch between gradual and normal, and do not suddenly override density and other settings. 

**Additional Information**

Anyone else?

filipgoc, 2018-05-25:
A related question: How do you thin I should provide a fallback infill density? 

I now tried this, 

```
infill_sparse_density = =77 if gradual_infill_steps > 0 else 33
```

but it does not react. stays 77 all the time....

ianpaschal, 2018-05-26:
I'm not sure about this at all, but I have a suspicion that checking gradual infill applies more than the gradual infill setting but also sets the density to 90 so there's an obvious shift. In theory checking gradual infill should require two points on the slider which to grade between.

fieldOfView, 2018-05-26:
It looks like the qml tries to remember the previous infill density setting, but fails to remember a setting set in the profile/definition.

filipgoc, 2018-05-26:
I'm leaning towards _design flaw_ rather than a _bug_: 

1. If I understand correctly, _gradual infill has no enabled/disabled state separate from its settings_. This is the real culprit. Since the settings determine whether it's enabled or disabled, it's hard to keep fallback in check. 

```  
"gradual_infill_steps":
                {"enabled": "infill_sparse_density > 0 and infill_pattern != 'cubicsubdiv' and not spaghetti_infill_enabled",
...

"gradual_infill_step_height":
                {"enabled": "infill_sparse_density > 0 and gradual_infill_steps > 0 and infill_pattern != 'cubicsubdiv'",
...
```

2. Then the question is, what does the 'enable/disable' toggle in the GUI do? 
    - (I don't know how to go digging about to figure this out. I never did any GUI. )

3. Lastly, where are the values flying in from?!

It resets from

```
gradual_infill_step_height = 1
gradual_infill_steps = 1
infill_sparse_density = 70
```

to
```
gradual_infill_step_height = 1
gradual_infill_steps = 5
infill_sparse_density = 90
```

And these are not fdmprinter.def.json defaults... where is this coming from?

ianpaschal, 2018-05-26:
I haven't checked but I'm pretty sure the answer to 2 & 3 is:

> It enables gradual infill and sets the percent to whatever is considered the "default" for gradual infill.

Since that isn't what one would expect, I think it's safe to call it a bug. Also we don't have a label for design flaws. 😛

fieldOfView, 2018-05-26:
The fallback value when there is no “remembered” value is hardcoded in qml

filipgoc, 2018-05-26:
> Also we don't have a label for design flaws.

HAHA 😈

filipgoc, 2018-05-26:
If it falls back onto 'default' values as provided in fdmprinter.def.json, then designers could provide gradual infill defaults as overrides in printer definitions... but this would still make it unpredictable to users who don't have access to the machine definitions. I'm curious how this bug will get squished. My humble suggestion would be to just add an explicit enable switch.

fieldOfView, 2018-05-26:
An enable switch would not fix the issue; there’s still a different recommended infill density with gradual infill enabled. 

With how settings work, a formula value that references gradual infill would be overwritten as soon as the user moves the infill slider to set a custom value.

filipgoc, 2018-05-27:
True. I was thinking you could use something like
```
infill_sparse_density = =77 if gradual_infill_enabled else 33
```
, but any GUI tweaks would would destroy that... 

-> New proposal (just trying to finish this up so next time noone needs to think again.)

```
gradual_infill_enabled
gradual_sparse_infill_density
```

There two should cover it. (I hope)

filipgoc, 2018-05-27:
PS: unlike now, the infill slider would _not_ get disabled when gradual infill is enabled then - it would just be changing a different variable in the background. the image to the right is then a good visual reinforcement that it's enabled now, so I think it works fine UX wise.

Ghostkeeper, 2018-07-11:
We've been discussing this and decided that we cannot gracefully fix this. We found two candidate solutions:
- Track two different infill densities, one for when gradual infill is enabled and one for when it's not enabled. This is Filipgoc's proposal too. There is a user experience problem with that though that the infill density setting could be modified in an old profile and have no effect any more because gradual infill is on. And that you might set the infill density to something, disable gradual infill, then a year later re-enable it and wonder why your infill density changed. Apart from that it also adds more complexity than we'd like.
- Instead of halving the infill density as you go down, double it from the base density as you go up. This is also a bit of a problem for existing profiles but would make the code base simpler in some ways. This way the base infill density can be the same for both types by default. There is a user experience problem with that too though, because when you then enable gradual infill your printing time will increase instead of decrease.

closed 2023-06-24 by github-actions[bot]