Ultimaker/Cura #3856: Add formula inputs for settings

ianpaschal, 2018-05-26:
This is a migration of a proposal by @awhiemstra posted in a Cura 2.4 beta bug. Since that issue is incredibly old and won't be addressed as a bug report, I closed it. It does make a nice "new feature" proposal though, so:

**@awhiemstra:**

>Adding a way to enter formulas in the UI has been on our wishlist for a while actually. It is however not as straightforward as just showing the formula, as the underlying formula actually uses the setting keys which are not exposed to the UI. So we need to come up with an editor widget which allows you to enter setting names and then maps that to setting keys. We also need to do some validation on the formula and a bunch of other things. Since we have more pressing concerns we never really got around to implemeting this.

Obviously this is a _big_ feature and difficult to build, but that's why I made it its own thread.

ChrisTerBeke, 2018-05-27:
If something like this were to be build, I'd like a 'instant' preview of the formula result while you're entering it.

Nerdy314159265, 2018-07-06:
This is something that I really, really want. It'd make some settings way easier to adjust.

For example: some people like to do their infill height at twice their layer height so it saves some time and the thicker layers make it stronger.

konskarm, 2020-11-10:
We rediscussed this feature again today and we estimated it to be a very large change that we will be used that often. Therefore, we will defer it due to the amout of work needed to achieve the desired result.

closed 2023-07-13 by github-actions[bot]