Ultimaker/Cura #3858: anycubic i3 Mega wird bei Programmstart nicht gefunden

fbsln, 2018-05-27:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

??? Wer kann helfen ???

Danke

ChrisTerBeke, 2018-05-27:
English please.

Ghostkeeper, 2018-07-06:
Translated this to English.

Ghostkeeper, 2018-07-06:
If the entire side bar is gone, try pressing Ctrl+E to get it back. It is a toggle that you can press to get more screen space for your model.

If the side bar is there but is empty and white, I think you have a problem with rendering. We don't display "Compatibility mode" as separate text anywhere (only in the preferences menu regarding layer view). I think something from your computer is putting that there.

jackha, 2018-07-16:
If it says compatibility mode out of nothing, it may have to do with your gpu. Did you install the newest video card drivers?

If you don't mind your old settings, you can also delete your config folder to see if that helps.

Ghostkeeper, 2018-08-28:
I think we'll have to assume that Ctrl+E or updating video drivers fixed the problem then.

closed 2018-08-28 by Ghostkeeper