Ultimaker/Cura #3859: Makerbot Replicator 2 Noise

lopesff, 2018-05-28:
Hello,
I´m trying to use CURA in my Makerbot Replicator 2 but, I can not. When I slice my part and create a x3g file I put sd card in my printer but doesn´t work.
My printer only make some noise and doesn´t start to print.

How can I fix it?

ianpaschal, 2018-05-28:
This sounds like it's likely an issue with your hardware, rather than Cura. Also we can't provide technical assistance for Makerbot printers.

That being said, it's impossible to say for sure if it's a Cura issue without filling out the issue template and uploading some screen shots, models, test projects, etc.

Please fill that out or this ticket will be closed.

```
<!--
The following template is useful for filing new issues. Processing an issue will go much faster when this is filled out, and issues which do not use this template WILL BE REMOVED.

Before filing, PLEASE check if the issue already exists (either open or closed) by using the search bar on the issues page. If it does, comment there. Even if it's closed, we can reopen it based on your comment.

Also, please note the application version in the title of the issue. For example: "[3.2.1] Cannot connect to 3rd-party printer". Please do not write things like "Request:" or "[BUG]" in the title; this is what labels are for.

It is also helpful to attach a project (.3mf or .curaproject) file and Cura log file so we can debug issues quicker.
Information about how to find the log file can be found at https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations. To upload a project, try changing the extension to e.g. .curaproject.3mf.zip so that github accepts uploading the file. Otherwise we recommend http://wetransfer.com, but other file hosts like Google Drive or Dropbox work well too.

Thank you for using Cura!
-->

**Application Version**
<!-- The version of the application this issue occurs with -->

**Platform**
<!-- Information about the operating system the issue occurs on -->

**Printer**
<!-- Which printer was selected in Cura. Please attach project file as .curaproject.3mf.zip -->

**Steps to Reproduce**
<!-- Add the steps needed that lead up to the issue -->

**Actual Results**
<!-- What happens after the above steps have been followed -->

**Expected results**
<!-- What should happen after the above steps have been followed -->

**Additional Information**
<!-- Extra information relevant to the issue, like screenshots -->
```

lopesff, 2018-05-28:
​I can use makerware or S3D and I don´t have problems.
My printer is a Makerbot Replicator 2
I select at CURA Makerbot and GCODE flavor I put Makerbot.

I saw a youtube video with the same problem
https://www.youtube.com/watch?v=dtHib1chdoI

But my makerbot not home axes neither start to print.
Could you help.

I´m using the last release CURA.​

ianpaschal, 2018-05-28:
> Please fill that out or this ticket will be closed.

closed 2018-05-28 by ianpaschal

Ghostkeeper, 2018-07-09:
I think this was an issue with the X3GWriter that it didn't get the proper settings for your printer. I have an update ready for release that would solve it. Please re-download the X3GWriter in a few days and give it another try.