Ultimaker/Cura #3860: [3.3.1] USB monitor/control is just too spartan.  Where is everything?

VanessaE, 2018-05-28:
Basically, compare Cura's USB montior area:

[image]

Pretty spartan, honestly.  Now Pronterface, just after resetting the printer, just to make its startup spew visible in the debug console area:

[image]

I'm not saying that Cura needs to duplicate this, but its print-over-USB is so minimal that Pronterface basically just eats Cura's lunch (and breakfast, and dinner) here. :smile:

**Application Version**

3.3.1

**Platform**

Debian 9.3

VanessaE, 2018-05-28:
I meant to add a couple of things:  Cura's abort print button should NOT execute some hard-coded set of instructions (because, among other things, it turns off my Z motors, which I do NOT want).  If anything, it should execute the "end" g-code in the printer profile.

When paused, none of the jog controls do anything.  They should only be locked out while actively printing.

The send-command box does not respond to keypad "enter", just the main "enter" key.  This is habit for most people, simply because the majority of g-code commands end with a number, so your hand is already on the keypad (well one assumes).

There's no echoback/debug console at all in Cura.  How do I know what the responses were to say `M114`, `M119`, or `M851`?

When printing, Pronterface tracks the current print line by line in realtime graphically.  Cura has its "playback current layer" facility provided by Cura (you know, the "▸" button above the layer select slider, where a ghost of a hotend is made visible).  I think that facility should be duplicated and used to track the ongoing print (so that you don't lose the ability to go off and work on a new print while the current print is going).

I'm sure a few things in Pronterface's UI may not apply to Cura, but you get the point.

Pronterface, to me, represents a fairly modern, full-function printer control interface (ironically, some people still consider it outdated.  go figure), but as much as I like it, I'd really prefer not to have to run two programs to manage my printer.

ChrisTerBeke, 2018-05-28:
There's like 10+ other issues here about USB support. Please have a look at those first and see if your comments were already replied to before. In short, we don't really support USB from the Ultimaker team because our printers don't support it. If the community wants to improve USB control in Cura, that's fine.

closed 2018-05-28 by ChrisTerBeke