Ultimaker/Cura #3862: 3.3.1 - Initial Infill Path

amagr0, 2018-05-28:
**Application Version** - CURA 3.3.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Generate the gcode and in "Layer View" analyze the toolpath.

**Actual Results** - The path of the Infill starts far away from the last wall path extrude.

**Expected results** - Infill path extrude starts near as possible from the last wall extrude.

[image]

ChrisTerBeke, 2018-05-28:
My guess is that it selects a starting point that results in the largest single line to print the infill so no extra retractions are needed.

smartavionics, 2018-05-29:
It's not possible from a single image to understand the ordering used but, in general, the infill writer tries to output "chains" of lines. Two line segments are chained if one starts very close (within um) of the other ending.  Having finished one chain it then finds the nearest start to another chain. It does not attempt to order the chains to minimise the overall distance travelled. If you post a project file, I can comment in more detail on your example.

amagr0, 2018-05-29:
[cube50.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2047375/cube50.curaproject.3mf.zip)

smartavionics, 2018-05-29:
Thanks for the project. I loaded it and without changing any settings got a result that doesn't look the same as your picture. All the infill layers look the same. The chains are being followed and the total amount of travel is small (2%).

[image]

dmitrygorbenko, 2018-05-29:
@smartavionics can you show where is the beginning of the first line on "infile" ? please, show us a start point in a way like @amagr0 did.

I have many concerns about the same problem -- Cura (3.3.1) does a lot of movements instead of printing. Maybe it's a 2%, like you wrote, but these movements don't make to much sense to me. Why not to start extruding as near as possible ? Why the nozzle moves to other side of the wall and continues printing there instead of continue printing right here next to the place where is has just stopped ?

smartavionics, 2018-05-29:
Here is an image showing the first infill line, you can see that it is close to the corner where the walls start/finish.

[image]

Cura 3.3.1 has a horrible bug in the combing that makes it travel "the long way round" although that doesn't appear to be the problem in the OP's image. There have been various other changes so perhaps it has been fixed for the next release?

amagr0, 2018-05-29:
This is the same exact layer (276), and it's not starting as the previous image. 
[image]

smartavionics, 2018-05-29:
Something must be different in the engine. I am using a derivative of the master branch so the change that has been made will probably be in 3.4 (along with the latest selection of bugs!)

amagr0, 2018-05-29:
Ok, it's explained!
Is there already a release date for the 3.4 ?

smartavionics, 2018-05-29:
Sorry, I don't know but I noticed today that 3.4 branches have appeared in the Cura repos so I guess there will be a beta available in a few weeks from now.

amagr0, 2018-06-06:
It seems that was not solved in 3.4 beta.

[image]

smartavionics, 2018-06-06:
That's odd because it's solved in a 3.4 related release that I have here:

[image]

amagr0, 2018-06-07:
Can anyone else confirm this issue was solved on 3.4 beta?

jackha, 2018-07-16:
In the current master, I see exactly what @smartavionics also sees. I hope it also does that for @amagr0 because otherwise we can't reproduce it and thus fix it.

[image]

closed 2018-07-16 by jackha