Ultimaker/Cura #3864: Remove Compatibility Mode and provide full support for old OpenGL version like Simplify3D does

dmitrygorbenko, 2018-05-29:
I don't have on my linux system the latest OpenGL version and because of this I can preview gcode in compatibility mode.

I know what prior some version (like 3.0.0 or earlier) I was able to use full-graphic preview on my machine, but now I can't.

Simplify3D doesn't complain, it just renders all stuff and don't break graphics.

Please, remove "Compatibility Mode" in preview mode and rewrite your code in "Normal mode" to support "old" OpenGL versions.

ChrisTerBeke, 2018-05-29:
Pretty sure this is not gonna happen. We just need OpenGL 4.1 for certain operations and rendering.

ChrisTerBeke, 2018-05-29:
Also, this requirement has been there since Cura 2, so something else on your system must have changed for it to suddenly not work anymore.

closed 2018-05-29 by ChrisTerBeke

Ghostkeeper, 2018-07-12:
The necessity for OpenGL 4 comes from that we use a tesselation shader to generate the 3D tubes for the more modern layer view. We could do that with a pure geometry shader in OpenGL 3.2 but it will still require a compatibility view to support OpenGL 2.

awhiemstra, 2018-07-12:
Actually, it already uses a geometry shader, there are no tesselation shaders involved (unless that changed recently). The reason we use OpenGL 4 is simply because PyQt does not expose the OpenGL 3 version of the OpenGL bindings, just 2.1 and 4.