Ultimaker/Cura #3865: [3.3.1] extrusion to low if “inner wall(s) line width” > Outer

esopalumpa, 2018-05-29:
**Application Version**
3.3.1

**Platform**
linux amd64

**Printer**
BCN3D Sigma

**Problem**
if `wall_line_width_x >  wall_line_width` (_Inner Wall(s) Line Width_ > _Outer Wall Line Width_), the extrusion-calculation seems to be wrong.
The lines seem to be spaced according to wall_line_width_x but extrusion is too low, resulting in gaps between walls.

Ghostkeeper, 2018-07-12:
I could reproduce this. Strange thing though: Layer view indicates the correct line width. Something is going wrong with the extrusion calculation.

jackha, 2018-07-16:
Looked at it again with @Ghostkeeper and we can't reproduce it. 
@esopalumpa can you try 3.4.1 and see if it's fixed? If it's not fixed, can you provide us with a project file or so where this occurs?

esopalumpa, 2018-07-17:
In this example the outer wall is 0.4 mm and the inner wall and the filling are 0.8 mm. (The object in the center is **one** part.)
As you can see the inner wall has the same width then the outer wall (or even less?) while spacing is for a wider wall.

[image]

[CFDMP_Cura-test.curaproject.3mf.gz](https://github.com/Ultimaker/Cura/files/2200531/CFDMP_Cura-test.curaproject.3mf.gz)

esopalumpa, 2018-07-17:
Btw. It is not the fault of the printer. If printing with Slic3r with equivalent settings (as far as I could get it) it works much better with the exception of a slight overextrusion.
Settings: 0.48mm / 0.8mm.

[image]
:

jackha, 2018-07-18:
Really strange. If I look at the g-code interpretation (that calculates the expected width from the flow), then it looks good. I also haven't heard this problem from others. Does it look good on your computer as well?

One last thing to check: are the nozzle size and material diameter set correctly? 0.4 / 2.85

[image]

esopalumpa, 2018-07-18:
It looks the same on my computer. Nozzle sIze and filament diameter are O.K.
I took a look into the generated g-codes and as far as I can see the extruded amount of filament should be nearly the same in the Cuda and Slic3r code (difference ±6%).

I have no idea why the cuda-generated code does not work as intended on the printer.