Ultimaker/Cura #3867: Feature Request - Minimal Extrusion Distance

amagr0, 2018-05-30:
In order to avoid some really small extrusions, Cura could have a setting that will define the minimal extrusion distance.
This could speed the print and avoid a lot of retractions and some cloggings.

For example, in the image below I choose 3 perimeters, but as you can see the most interior perimeter extrusion on the columns are so small that could be dispensed.
[image]

And also on this image:
[image]

smartavionics, 2018-05-31:
Perhaps you are thinking of the Min Wall Flow setting that has just been merged to master? That does something else, it omits wall segments whose flow is below a threshold. The primary use for it is to avoid doing the nearly (or completely) empty extrusions that occur when a wall has been flow reduced a lot to fit it into a small gap.

amagr0, 2018-06-01:
"omits wall segments whose flow is below a threshold" this is similar to the feature that I suggest. And please don't forget to add this feature also to the infill.

Ghostkeeper, 2018-07-13:
You think that the "Filter Out Tiny Gaps" feature should be extended to filter out tiny walls as well? It's sort of the same extrusion.

LipuFei, 2018-07-13:
JIRA ticket: CURA-5571