Ultimaker/Cura #3868: [3.3.1] "Gradual infill step" should ignore "alternate extra walls"

Zer0t3ch, 2018-05-30:
**Application Version**
3.3.1

**Platform**
Windows 10 x64 build 1803

**Printer**
[CFDMP_Sensor_Mount_v2z.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2054129/CFDMP_Sensor_Mount_v2z.curaproject.3mf.zip)

**Steps to Reproduce**
Enable "alternate extra wall" and set "gradual infill steps" to anything more than one

**Actual Results**
The "extra" infill is used on the edge in-between each extra wall, causing a lot of unnecessary jerking around.

**Expected results**
The space between extra walls should just be using the "normal" infill percentage.

**Screenshot** 
This should reasonably show what I'm talking about.  
[image]

ChrisTerBeke, 2018-05-31:
Seems like something to improve indeed.

Ghostkeeper, 2018-07-13:
We acknowledge that this is a problem, but we'll put it on low priority since the extra wall alternating isn't used very often. Maybe this'll get a look when we debug the other problem that we have right now with high infill overlap in combination with gradual infill. We'll put this off for now.

m-sundman, 2020-12-31:
IMHO this is not a feature request. The 2 features clearly conflict with each other nonsensically, so it should be classified as a bug.

(And what do you mean _"extra wall alternating isn't used very often"_? Isn't everyone using it? A modest increase in print time gives you 90% of the benefits of an additional wall, and extremely good infill adhesion (better than connecting infill lines), so enabling it is a no-brainer!)

Ghostkeeper, 2021-01-08:
It's not behaving as desired. It's behaving as expected, since Gradual Infill is expected to increase the density towards the top of the infill volume. Since the infill volume is serrated, there are lots of little ledges at the side of the print. Therefore it would be a bug if this is not calculated correctly to include those little areas as being high infill density.
Being a feature request or a bug has nothing to do with how important an issue is. It is an issue, but behaving as expected of the software. Just a point that can be improved by making the expected behaviour something else.

> Isn't everyone using it?

No. In the records of the last month, 1.79% of slices had this setting enabled.

m-sundman, 2021-01-08:
> there are lots of little ledges at the side

Yes, I understand why it's doing what it's doing, but my point was that those particular ledges aren't really top or bottom, at least not conceptually. They might be implemented as such, or they might be implemented in some other way. Fixing this bug might be a matter of just swapping the order of the two operations in the processing pipeline, or it might require a lot more. Still the alternating walls are not there as tiny tops/bottoms, but as infill-grippers, so adding extra infill just inbetween them without reaching the infill goes against their very purpose.

> 1.79% of slices had this setting enabled

OK. That surely means that it's disabled by default, but I understand, the effect is still that the problem won't affect 98% of the users.

pixie-wrangler-irl, 2021-08-15:
This is still an active bug in version 4.9.  Sure would be handy to reduce infill travel time on long prints.