Ultimaker/Cura #3870: Cura 3.3.1 - "Brim only on outside" is ignored

Drayson80, 2018-05-31:
Seems there is a bug :-)
Sliced a flat, 1mm thick stencil with square outside and a castle-shape inside form.
Dont want a brim on the inside as it is horrible to clean.
Nevertheless, the function "Brim only on outside"  seems to be ignore - or have I misunderstood something about this feature...?

[image]

Any hint appreciated...

BagelOrb, 2018-05-31:
That's odd. Could you share your project file please?
(Rename to zip to upload in GitHub)

Drayson80, 2018-06-03:
[Tortendeko Castle_1.STL.zip](https://github.com/Ultimaker/Cura/files/2066242/Tortendeko.Castle_1.STL.zip)
@BagelOrb , here we go... attached the file, just rename it back to STL only.
Thanks for assistance...

Drayson80, 2018-06-03:
sorry, you mean the cura project file - isn't?
Here it is too...
[Tortendeko Castle_1.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2066250/Tortendeko.Castle_1.curaproject.3mf.zip)

BagelOrb, 2018-06-05:
Oh I see what the problem is.
There are several small parts inside the cutout which are not connected to the large part.

These small parts also need a brim on their outside, but because this brim would then intersect with the large part, the brim is continued along the inside of the hole in the large part.

I suggest you alter the 3D model such that all the parts are connected to each other with pieces of at least the nozzle size in width.

closed 2018-06-05 by BagelOrb

Drayson80, 2018-06-05:
AAAAaaaaahhhhhh..... dude, youre great, thank you for clarification. sorry, my fault...