Ultimaker/Cura #3871: Cura 3.3.1 not slicing. After hit "Prepare" button nothing happens.

Kikerumbo, 2018-05-31:
**Application Version**
Cura 3.3.1

**Platform**
Windows 10 x64

**Printer**
Ender 3 (Custom CR-10 profile)

[CCR10_Matek F405-Wing mount.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2058028/CCR10_Matek.F405-Wing.mount.curaproject.3mf.zip)

**Steps to Reproduce**
Hit "PREPARE" button

**Actual Results**
Nothing, it just shows the "Slicing" text, but the bar doesn't move. Waited for 2 hours and nothing happened

**Expected results**
Obtain a G-Code file to put in a SD card.

**Additional Information**
Tried put the Cura folder in the exception list of the firewall/antivirus, executing Cura as administrator, reinstalling it. Looked for solutions for older versions.

Cura.exe process is running, but not using CPU. No new process when I try to slice.

[cura_CFG.zip](https://github.com/Ultimaker/Cura/files/2058079/cura_CFG.zip)
[image]

Thanks.

diegopradogesto, 2018-06-01:
In my first attempt using 3.3.1 it slices correctly and instantly. I will take a look at the current configuration and I will come back to you.

ChrisTerBeke, 2018-06-03:
I think this has been addressed several times before on GitHub. It mostly turned out to be a permission issue with CuraEngine on Windows. Try running Cura as administrator?

Kikerumbo, 2018-06-03:
Thanks for your answers.

As told in my previous post, yes, I already tried to run Cura as administrator. Same behavior.

ChrisTerBeke, 2018-06-03:
MM weird. I also tried to open your 3MF project but Cura (3.3.1) just won't open it...

Kikerumbo, 2018-06-03:
I tried with different .stl files. Not one works. If I use other slicing software they seems to work just fine producing the G-code.

I suppose that when I hit the Prepare button CuraEngine.exe should start running, but I can only see the Cura.exe process in the task manager.

Any suggestion?

By the way, thanks for trying.

ChrisTerBeke, 2018-06-03:
Best thing I can still think of is a permission issue. Maybe a complete re-install (maybe on another drive even?) could fix something. If not, you could try the 3.4 Beta which is due next week.

diegopradogesto, 2018-06-04:
Probably is related with permissions as @ChrisTerBeke says. Maybe we can find something interesting in the logs. Can you provide the log file?

Alesky07, 2018-06-04:
To share my experience,
My Cura installation is on an old computer, and after clicking on prepare button I have to wait several seconds (sometimes 20 seconds) until it start to slice.
I did not have this lag on a recent computer.

diegopradogesto, 2018-07-10:
@Kikerumbo , are you still experiencing these problems? Have you tried with version 3.4? Is it maybe due to lagginess with your computer?

We need more info in order to address your issue.

Kikerumbo, 2018-07-10:
I no longer use CURA.

I do not have any lag with other slicing software (i5@4.6Ghz, 16Gb RAM and SSD).

diegopradogesto, 2018-07-10:
Ok, so... since nobody else is having this problem for now, I'll close this issue.

closed 2018-07-10 by diegopradogesto