Ultimaker/Cura #3872: Cura won't slice for custom FDM printer with PVA support

Evgaleks, 2018-05-31:
Unable to slice with **PVA support** when selected Custom FDM printer !!!

**Application Version**
Cura 3.3.1
**Platform**
Windows7 x 64

**Printer**
Custom Printer with dual extruder. 

**Steps to Reproduce**
Unable to slice with anyPrint setup  Recommended or Custom 

**Actual Results**
[image]

**Additional Information**
Extruder 1     nozzle 0.4  - **PLA**  1.75
Extruder 2     nozzle 0.4  - **PVA**  1.75

diegopradogesto, 2018-06-01:
The PVA was set as not compatible since it provokes clogs in the nozzle very easily. However I agree that is something that should be left to user decision.
As a workaround you can create a new PVA material in Preferences > Materials and then modify it manually: 

- Go to your user data folder (somewhere in C:\Users\<username>\AppData\Roaming\cura\3.3\materials)
- Remove the line marked with `<--` in the following example:

```
...
<settings>
     <setting key="hardware compatible">no</setting>   <--
     <setting key="print temperature">215</setting>
     <setting key="standby temperature">175</setting>
...
```

Thanks for the report.

diegopradogesto, 2018-06-04:
We created an internal ticket for this issue. It is now in progress. For developers: CURA-5444

Evgaleks, 2018-06-09:
The same problem occurs if, on the standard Ultimaker 3 printer, in the settings, replace the diameter of the filament by 1.75mm. All other configuration settings are identical.

Ultimaker3 with 1.75mm filament: (Unable to slice with PVA support) 
[image]

Ultimaker3 with 2.85mm filament: (Slicing with PVA support)
[image]

diegopradogesto, 2018-06-11:
The issue is now fixed and it will work on the 3.4 release that will come in a few weeks after the beta.
Thanks for the info.

closed 2018-06-11 by diegopradogesto