Ultimaker/Cura #3874: [All versions] Don't respond to background clicks

whosawhatsis, 2018-06-01:
**Application Version**
All versions (3.3.1 and earlier)

**Platform**
Mac (and possibly others).

**Steps to Reproduce**
Have Cura open with another app in the foreground. Click on the Cura window to bring it to the front.

**Actual Results**
Whatever interface element was under the mouse cursor will activate. If it's a menu, the menu will open. If it's a checkbox, (or the area to the right of a checkbox that was not visible behind the other window), the setting will change unexpectedly and re-slicing will commence.

**Expected results**
Clicking on the Cura window when it is in the background should only bring it to the front. Interacting with UI elements should require a second click, after the full window is visible.

**Additional Information**
Often, only the right-most portion of the Cura window will be visible, especially if large windows (like web browsers) are in the foreground. This right-side edge of the window has no safe areas to click to bring the Cura window to the foreground that will not be interacting with a UI element. In this state, the labels for these UI elements are likely covered up, making it impossible to know exactly what you will be effecting.

This is annoying when I click near the top of the window and open the printer or material selection menus, but it can be really damaging to prints when I click on a seemingly blank area that turns out to be the blank space to the right of a checkbox. When there is a checkbox there, the setting will be changed, but I often don't realize that it was changed until I try to change profiles and Cura asks me if I want to keep the accidentally changed setting.

I don't know about other platforms, but the Mac has UI features that allow a user to interact with UI elements of a background app (command + click), and when not using them, the default behavior should be to ignore the click and only bring the window to the foreground.

ChrisTerBeke, 2018-06-03:
This is actually an OS specific trait, and Qt (the cross-platform UI framework we use) apparently chooses to use this behaviour (probably because MacOS users are familiar with it). I've tried several other apps on my own Mac and most seem to use this behaviour as well, so for now we're gonna assume it's intended.

closed 2018-06-03 by ChrisTerBeke

Ghostkeeper, 2018-07-13:
This doesn't happen on Windows or Linux.