Ultimaker/Cura #3875: [3.3.1] Cura should never start a perimeter on an overhang

VanessaE, 2018-06-02:
I'm printing the vertical version of this part:

https://www.thingiverse.com/thing:2563185

As you can see, it consists of one horizontal beam, two vertical beams, and 45° diagonal beams forming an X.  This X, thus, has overhangs that face each other, when sliced.

No matter how I tell Cura to position the seam, one of the diagonal bits will have the seam placed right on one of its outermost corners (or too close to it).  For example, here is "user specified" mode with the target coordinates set to {100,600}, which inexplicably puts the right island's seam on the overhang:

[image]

Shortest has the same effect, the right island's seam starts on the overhang:

[image]

Sharpest Corner, the right island is fine now, but now the left island has its seam starting on the overhang:

[image]

I guess I don't need to show Random mode.

The print moves proceed counter-clockwise from the starting point (at least in Sharpest Corner mode).

The design of this part makes it impossible to rotate it to any orientation that would put all of the seams in desirable positions, because of the facing overhangs created by the X shape.

Basically, Cura should never -- in any seam placement mode -- start a perimeter on an overhanging corner, whether it's the exterior perimeter, or the corresponding interior ones.  I realize an interior perimeter is usually 100% supported by the lines under it, but it's still on an overhanging corner.  Although there is some plastic there, there's nowhere near enough thickness to ensure a rigid base for each new layer.

In the actual print, as expected, this causes the affected corner to curl, warping the whole diagonal bit after a while.  Warp it enough and the nozzle crashes into it, forcing it down.  Too much further, and something will crash into it that will knock the part off the plate (like fan ducts).  The bit with the seam placed properly is perfect (or as much so as my bot can do right now).  There's plenty of cooling, and I'm printing slow with this part (30 mm/s).

I've had this happen in other models, too, but overhangs usually end up being short enough to not create a severe problem.  The part linked above, however, pushes the issue to its limit (at least on my bot).

**Application Version**

3.3.1

**Platform**

Debian 9.4

**Printer**

Prusa i3 clone, but that's not relevant here.

**Steps to Reproduce**

Load the vertical YACS model from the above link, select any seam placement mode, and slice it.

**Actual Results**

One or another overhanging corner will get a seam.

**Expected results**

The seam should be started as far away from any overhanging corners as possible, and with as much printing distance as possible between the corrected start point and the first overhanging corner to be printed.

In the screenshots, the seam should be placed caddy-corner to where it actually is (for the affected bits), placing it on top of what amounts to dozens of layers of underlying plastic (hence a rigid base).

smartavionics, 2018-06-02:
Hi @VanessaE, I think you have a good point. In the bridge handling code, I avoided starting walls over air and I think a similar thing could be done for normal walls. I will look into it.

smartavionics, 2018-06-02:
Actually, the bridging settings help you here. Set the min bridge wall length to 1, the bridge wall max overhang to 20 and the bridge wall flow to 100. Bingo!

[image]

[image]

VanessaE, 2018-06-02:
Interesting workaround.  To get this part printed, I set the seam to random (so that at least when a seam gets placed wrong, there won't be many of them stacked up).

Since you brought it up, overhang settings should be distinct from bridging settings (flow, speed, and acceleration especially).  I guess it's okay to use a threshold to decide whether to consider something a bridge or an overhang, though.

smartavionics, 2018-06-02:
> Since you brought it up, overhang settings should be distinct from bridging settings (flow, speed, and acceleration especially). I guess it's okay to use a threshold to decide whether to consider something a bridge or an overhang, though.

Ideally, the overhang settings would be separate from the bridging settings but that would make the implementation even more complicated than it already is. At the moment, when printing a wall line, it has to segment the line into those portions that are above air and those that are not and then it does some funky stuff with the extruder flow rate to try and avoid over/underextrusion due to the step change in extruder rate that is required when crossing between bridge/non-bridge.

Adding separate params for the regions of the line that are considered overhang would be nice but how many models have both bridges and overhang?

VanessaE, 2018-06-02:
Good points, all.

"How many models"?  Honestly, a lot of them.  Particularly anything containing a non-rectangular horizontal hole (e.g. for a screw/bolt), as the upper half of the hole is of course all overhangs, and the top layer closing the hole is formed by bridging.  Anything mechanical and remotely complex is almost guaranteed to have both features.

My most recent important print was a replacement for the trunk pull-down gearbox for my car (https://www.thingiverse.com/thing:2929763 in case you're curious).  On first look, you'd almost think it has more overhangs and bridges than it has regular vertical walls. :wink:

AbeFM, 2018-06-02:
Totally agree here, as a general rule, start extrusion where they are most likely to work. There used to be similar issues with a tower coming from a flat surface - printing its base on gaps in infill.

ChrisTerBeke, 2018-06-03:
Good suggestion indeed, it'll improve print quality for overhangs.

VanessaE, 2018-06-15:
Beta 3.4.0 is out (as you surely know); is there any chance that fixes for this seam placement issue (and maybe the part about overhang vs. bridge settings) can make it into 3.4.0-release?

ChrisTerBeke, 2018-06-15:
We almost never add features between a beta and stable release, only the necessary bug fixes.

VanessaE, 2018-06-18:
I'd define this as a bug.  Starting a seam on an overhang can render something unprintable if the overhang has something else running alongside but separate from it (as in many articulated "print in place" models).

ChrisTerBeke, 2018-06-18:
It's been there forever, so it's not a pressing bug for us.

VanessaE, 2018-06-18:
There are many things in this world that have "been there forever".  That doesn't mean they don't need solved.

Maybe it's not pressing to you, but what about to your userbase?  If I've run into this, you can be sure others have, too.  The usual 100:1 rule applies.

ChrisTerBeke, 2018-06-18:
I'm not here to debate priorities, that's something that the team and Ultimaker decide. If anyone else finds it a priority, they can always contribute. Let's just say that there's a list of other stuff that to work on that we deem more important at the moment.

VanessaE, 2018-06-18:
Fair enough, but with all due respect, I've skimmed through the issue list just in the past couple days (just looking for issues that I may have run into), and this is not the first time I've seen this kind of reply, and to be blunt, it reflects badly on the project, the company, and the individual contributors.

AbeFM, 2018-06-18:
I tend to agree with her. The responses to "I can't print this reasonable
thing due to an obvious problem with the approach" I've seen on here often
fail to reflect strong a commitment to improving to software for the user
base.

I understand not all these outsides can be handled right away, but the
responses aren't encouraging.

A suggestion on how to get around the problem without patching the software
would be good. Of the issue were not knowing how to implement the fix, we
could work on it here - I have written a lot of forum pseudo-code over the
years and real programmers have implemented it. I'm sure the fine folks
here can look at it.

This isn't meant to be telling anyone off - when I engage in conversation
here it is to try to help find a solution for the wider community.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

smartavionics, 2018-06-18:
Realistically, in the short term, the only way this is going to get developed is if I put the time in as the UM folks have a different set of priorities to the rest of us. I'm not adverse to doing that but as it's not a burning issue for me I will do it in my own time. As there are already 2 outstanding PRs of mine related to the bridging. I want to get them through the system before I create further mayhem.

smartavionics, 2018-06-18:
>  and to be blunt, it reflects badly on the project, the company, and the individual contributors.

Really? Well that sends you to the bottom of my list of people to make happy by contributing my own time for no material reward.

AbeFM, 2018-06-18:
Thanks for that. We're not aware of the behind the scenes stuff, and it's
hard to differentiate being too busy from not agreeing that an issue is
worthy.

I'd be quite happy to see this "on the list". Thanks again for all the
improvements so far.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

ChrisTerBeke, 2018-06-18:
> it reflects badly on the project, the company, and the individual contributors

I can also not respond and leave everyone in the dark. Just trying to be clear that it's not a priority for the people that are paid by Ultimaker at the moment. But since Cura is open source, that doesn't mean that it's not a priority for others that contribute to the project.

We shouldn't underestimate both the major open source contributors nor the funds that Ultimaker puts into this. I think it's only fair that both groups decide on their priorities. When those align, that's great. But sometimes they don't and we can still get a lot of work done between all of us.

VanessaE, 2018-06-18:
@smartavionics, "contributors" in this context should have clearly meant "UM employees and paid contractors", not (presumably-)unpaid people like you and me.  Don't have time?  Fine.  I get that.  Next time, just say it.  You do not have to get nasty.

Besides, this is 3d printing... digital model in, plastic out...  it doesn't get any more "material reward" than that. :stuck_out_tongue_closed_eyes: 

@ChrisTerBeke I don't reject the notion of priorities, but all too often, those priorities just are not well-considered.  I see it frequently.  Bottom line, unless there are viable workarounds (that don't involve jumping through ridiculous hoops), "broken for everyone" needs to be put above "broken just on our line of printers".  "Costs our users money" (for a free-as-in-beer project) needs to be put above all else.

Guys, I am struggling to be polite here, because frankly, these replies piss me off.  So I apologize for that air.

ChrisTerBeke, 2018-06-18:
> those priorities just are not well-considered

In your opinion I assume? There is no way to determine that objectively. Our priorities are also just opinions, they just happen to be the opinions of the people paying for the development. There's a lot more going on behind the scenes than just 'is it broken?'. We have to deal with a great amount of internal priorities from other departments. Which is perfectly normal because Ultimaker is a for-profit company.

I get that it's not an ideal situation, but on the other hand, without Ultimaker paying for a lot of this development, Cura might not even be here at all, or at least not get a new release very 10 weeks.

Note that I'm currently not in the core team of Cura in terms of daily development, but as someone who worked on it daily for some time, and also did some PRs in my spare time, I feel the need to explain that there's more behind Cura's development priorities than only looking at what is assumed to be broken.

VanessaE, 2018-06-18:
Indeed, it's just my opinion.  Don't get me wrong, I don't mean to devalue the work done by paid employees.

But, surely there are more than just you two guys involved in this project (paid or not)...

As for "broken", what else can I say about a function that's not working properly, to the point that it fails an entire print?

AbeFM, 2018-06-18:
> function that's not working properly, to the point that it fails an
entire print?

This is getting pretty far afield, but that last bit is why I report stuff.

If it seems like it'll ruin a print and it's fixable, I bring it up.
Sometimes the responses don't make it clear of the issue is credible.

A discussion of the issue is reasonable even if implementing a fix might
not be immediate.

I hope we can all continue to use this approach. Adding tags like
"suggestion" or "bug" can go a long way towards making a poster feel like
their input is not being ignored. If the issue is trivial, then a response
can illustrate.

Perhaps further meta-discussions could happen elsewhere, let's return to
the original topic here!

 There's a general problem with individual extrusions beginning in mid air
that this is a case of. How can we fix it?
                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

smartavionics, 2018-06-19:
Just letting you folks know that I have separated out the overhang support from the bridging and there will be two new settings, one is the threshold angle which determines if a wall is overhanging or not and the other is a speed factor that allows the overhanging wall to be printed slower if desired. It also moves the wall start points to avoid starting a wall polygon at an overhanging point.

The code is new and will obviously require testing and improving before submitting as PRs. However, I am not submitting the PRs until the log jam of outstanding PRs goes away as there is some dependency on the earlier changes being accepted.

AbeFM, 2018-06-19:
That rocks! Can't wait to try it... And yet I can.

Thanks again, really curious to see the effect. Maybe slice that big X from
earlier in the thread?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

smartavionics, 2018-06-20:
Here's a couple of images that show how the start vertex gets moved so that it is not above air:

[image]

[image]

If you want to try this out (linux only) you can find a tarball at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

This is the current master branch with my additions/tweaks (YMMV).

smartavionics, 2018-06-20:
By default, the overhang wall angle is set to 90 to disable the feature and the overhang wall speed is set to 100%. You will need to adjust those values to suit your requirements.

gravity-addiction, 2018-07-08:
So far it looks to be printing the shell wall at the reduced speed and the inner walls at regular speed.

smartavionics, 2018-07-08:
> So far it looks to be printing the shell wall at the reduced speed and the inner walls at regular speed.

Yes, that's what it does. Is that a problem? After all, if the inner wall is overhung, the outer wall is going to be in outer space! (not supported at all).

gravity-addiction, 2018-07-08:
that is correct, they would be in the air. The problem is how the material lays down at the different speeds.

smartavionics, 2018-07-08:
> The problem is how the material lays down at the different speeds.

Well, you don't have to print the overhung wall at a different speed from normal, that's an option. You could elect to print the overhung wall at normal speed.

gravity-addiction, 2018-07-08:
OR, the path i've choosen is to bring up your mb-improve-overhanging-walls-dev branch of code and poke around till I fix it.

smartavionics, 2018-07-08:
> OR, the path i've choosen is to bring up your mb-improve-overhanging-walls-dev branch of code and poke around till I fix it.

Fix what? What's not to your liking?

gravity-addiction, 2018-07-08:
That it doesn't print overhangs without deforming the part.

smartavionics, 2018-07-08:
Err, yeah, right. Well, when you have it fixed up to your satisfaction, submit a new PR.

gravity-addiction, 2018-07-08:
Get some sleep dude, you seem on edge. Just here to throw some effort into the common goal.

I'm printing some control pieces right now. I still have to figure out some of the curaengine utils and function routines.

smartavionics, 2018-07-08:
No, really, I am serious, if you have any suggestions/fixes that improve how overhung walls are printed, then submit a PR and everyone will be able to benefit from your efforts. But, yes, I need to sleep....

smartavionics, 2018-07-10:
Hi @gravity-addiction , I have looked at your PR and I can see what you have done there. I don't think I want it to work like that because I would rather not change the wall speed for the regions that are not overhung. You know how when you change the wall speed, the surface finish changes and you get obvious banding. That's why if I want a super consistent wall finish I don't let the cooling logic slow the layers down.

If and when my PR gets accepted into Cura  I recommend that you either submit your PR to the Cura devs and let them decide whether they want to use it (it's not my decision) or just do what I have done for quite a few of my own PRs that they have not accepted and just keep using it in your own version of Cura. Good luck!

gravity-addiction, 2018-07-10:
it's funny you say changing the wall speed changes the surface. My machine sounds like it has a nerve disease when I print with your overhang logic, and doesn't produce a very good result.

[image]

smartavionics, 2018-07-11:
> it's funny you say changing the wall speed changes the surface. My machine sounds like it has a nerve disease when I print with your overhang logic, and doesn't produce a very good result.

I don't know what you mean by that but I would be interested to look at the gcode that is being produced. Could you please attach the gcode or even better, the cura project for the above so that I can see what is happening. Thanks.

jackha, 2018-07-11:
Hi @VanessaE . Thanks for creating the issue. We've discussed this in the team and we think this would be a very nice fix. However, it would take a considerable amount of time to implement this, so we decided not to and spend time somewhere else. I hope you understand.

smartavionics, 2018-07-11:
Err, @jackha , I have already come up with a fix but I haven't submitted a PR yet because I am waiting for my backlog of PRs to be accepted/rejected first.

jackha, 2018-07-11:
@smartavionics we're very welcome to! we'll see it coming then :-)

GreyArea1966, 2018-07-12:
I started a separate thread as I didn’t know this one existed.

There’s a good tes5 piece for overhanf

rburema, 2018-09-05:
Hello @GreyArea1966,
Duplicates happen :-)
Perhaps it's a good idea to put in a link to the particular model you want us to have a look at; I tried to search with those terms (and a few variations) on Thingiverse, but I couldn't find anything that jumped out to me as 'this is clearly the one model that they meant'.
cheers, Remco.

GreyArea1966, 2018-09-05:
If I’d typed “spring without support” it might have helped...it’s this one...

https://www.thingiverse.com/thing:687668

I find it a good test piece because the layers are essentially identical in area, just rotated to produce the spring. Drop it into Cura and step through the layers...most will start in a “sensible” position...until Cura decides on sone of them that it’s going to start the layer in mid air...

It shouldn’t need a special set of rules to handle that...it should just always...ALWAYS...find another start point...in order of preference

1. Above material from the previous layer
2. As close as possible to material from the previous layer
3. In fact, since it calculates in advance, even an option that it CREATES material on the previous layer...or at least asks the user if it should
4. As a last resort, tell the user that the slice contains an “unsupported layer start” or similar term...and an abort/retry/ignore type option...

rburema, 2018-09-06:
@GreyArea1966 Thank you for the link, it does seem like a decent test piece, especially if also mirrored, squeezed, etc.

We're aware of what the problem is. The reason I asked is because I was looking at @smartavionics pull-request(s), which seem to fix the problem (or at least the first of your points), and I wanted to test a few things on a model that wasn't part of the original report. I've used another model instead for now.

AbeFM, 2018-09-06:
> Hi @smartavionics <https://github.com/smartavionics>,
>
> Thank you for your work & attention to this matter.
>
> As you've probably noticed, we've merged the resultant pull-requests into
> master, awaiting the final OK from Q&A. On the whole it seems to us that
> the fix does work as intended, and the default-settings don't change the
> behaviour already there, so that's good :-)
>
> There's a minor issue in that we think that the setting-name 'Overhang
> Wall Angle' and its detailed description don't completely convey their
> intended meaning to the user. The crux of the issue is that 'Overhang'
> and/or 'printed using overhanging wall settings' could mean a lot of
> different things, depending on context, especially when that context is
> just 'Experimental'. For instance, some skin is also affected by this
> change. As another example, the concept of overhang is also important in
> the generation of support, but these settings don't do anything for that.
>
> If you've got any suggestions, please don't hesitate to share them :-)
>
> cheers,
> Remco.

Avoid Free Air Extrusions?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

smartavionics, 2018-09-06:
Hi @rburema, @AbeFM. I don't really mind if the settings get renamed so go ahead and call them what you like but bear in mind that the Overhanging Wall Angle setting does more than just help avoid free air extrusions as it also enables the overhung wall regions to (optionally) be  printed at a different speed. So to me I still think of it as the setting that control which walls are considered to be overhanging rather than enabling a specific function.

AbeFM, 2018-09-06:
I'm not against either naming scheme.

Call me anything you'd like, but don't call me late for dinner.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Vandrasc, 2018-09-07:
We decided to leave it as it is, the settings will be available in the next release. 

Thank you all for your feedback and contribution.

closed 2018-09-07 by Vandrasc

AdamBWillis, 2018-10-17:
@smartavionics, thank you for your efforts in fixing this.  I did not think I would be able to overcome this with Cura.  You are appreciated.

smartavionics, 2018-10-17:
You're welcome. Sorry it took so long to get implemented.

nemphys, 2018-11-15:
This new feature is huge, @smartavionics you nailed it!
Do you think it would be possible to also implement a "Overhang Wall Fan Speed" setting? It would be nice to be able to blow some more air while the overhangs are being printed.

smartavionics, 2018-11-15:
Hi @nemphys , I'm glad that you like it. Sorry, I have stopped contributing new features to Cura, I'm just doing bug fixes and "customer support".

chanders, 2018-12-08:
I signed in just to say thank you for this @smartavionics

smartavionics, 2018-12-08:
Thanks @chanders, happy printing!

AbeFM, 2019-12-20:
Me too, while fan speed on slopes wasn't my biggest concern, these settings
have helped a lot in making arbitrary shapes come out well.