Ultimaker/Cura #3877: Enable "Fill gaps between walls" only for skin layers

lokster, 2018-06-03:
Cura 3.3.1 Feature request. After a couple of months of printing with the new Fill gaps between walls / Filter out tiny gaps settings from #2665, I realized it might be more effective if there is another setting for filling the gaps - specifying on which layers you want the gaps to be filled.
E.g. top surface skin / bottom surface skin / everywhere (this can be incorporated in the existing dropdown) OR top X layers / bottom X layers / everywhere.
The fill gaps setting is mostly useful when printing objects with flat surfaces (like the keychain examples in #2665), so it makes no sense to waste time (& tiny bit of material) filling the gaps on hidden layers that will never be visible.

jackha, 2018-07-11:
Hi @lokster . Thanks for the feature request. We discussed this in the team and we decided not to implement this.
- It would probably result in better prints because you have less travels
- It probably would not save much time
- It introduces new settings which we try to prevent
- Is it all work the effort? We think the total effect is too small.

closed 2018-07-11 by jackha

an-tech, 2019-08-19:
I print a lot of long aluminium profile prototypes. The gap fill function only for skin would be very useful.