Ultimaker/Cura #3881: No plugin works Cura 3.3.1

alicemouse, 2018-06-03:
**Application Version**
Cura 3.3.1

**Platform**
windows 10

**Printer**
Custom FDM printer (Flyingbear p902)

**Steps to Reproduce**
In the extension tab, you can not select any plug-ins. When you hover the cursor, they do not respond and do not click.

**Actual Results**
Nothing

**Expected results**
The plugin should be selected and opened. Specifically, the model's auto-orientation

**Additional Information**
[image]

ChrisTerBeke, 2018-06-03:
Seems more like a UI issue than the plugins themselves not working (otherwise they would not appear in the menu). Do other menu items work properly for you?

alicemouse, 2018-06-03:
Now checked and yes, the entire interface of the top menu does not define additional sub-items.

ChrisTerBeke, 2018-06-03:
Could you try switching your Cura language to English and try again? (trying to exclude sources here, we've had some issues with Russian rendering before).

alicemouse, 2018-06-03:
I tried, but it did not work.

alicemouse, 2018-06-03:
Sometimes all work, but I haven't idea on what it depends

Ghostkeeper, 2018-07-13:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/1940.

Ghostkeeper, 2018-07-13:
In particular, RegenboogViske has found an apparent workaround: Reset the registry key `/HKEY_CURRENT_USER/Control Panel/Desktop/MenuShowDelay` to its default value of 400.

closed 2018-07-13 by Ghostkeeper