Ultimaker/Cura #3882: 3.3.1 - Slots in center of area - can see completely through part.

KJEJ, 2018-06-03:
When slicing parts with flat surfaces, the slicer leaves a thin gap in the center of the part on both the top and bottom layer.  You can see through the part from top to bottom.  The infill does cross this area.  Changing the width of the part slightly can make it go away.  Shows on top and bottom layer view.

I've tried the setting changes that were suggested in some other posts, but they have no effect.

This is the view showing the gap:

[image]

Here is an example .stl file that produced the above .jpg, zipped:

[v6 mm.zip](https://github.com/Ultimaker/Cura/files/2066343/v6.mm.zip)

I believe I am using the same settings as with previous versions (defaults, I believe).  I had a friend try it on his version of Cura 3.3.1 and he got the same problem.

**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Creality CR-10, but problem shows in layer view before printing

AbeFM, 2018-06-04:
This has been around a while. Perhaps it'll get some attention, but in the meantime, just don't use concentric, or at least use "lines" as inner layers to hide/fill gaps

jackha, 2018-07-10:
Hi @KJEJ and @AbeFM . I can happily announce that this problem has been addressed and it's fixed in the next upcoming release (it didn't make it to 3.4.1). The problem only applies to top/bottom pattern concentric. (CURA-5276)

closed 2018-07-10 by jackha

AbeFM, 2018-07-10:
Awesome news, looking forward to trying it!