Ultimaker/Cura #3885: not start after pause

askmix, 2018-06-04:
1. Сura 3.3.1 (ru language) do not start after pause, firmware marlin 1.1.8, win10 64bit. Сonstantly shows the temperature of the bed and hotend on 1 degree Celsius below the target. 2. With a quick double-click on the navigation buttons does not move a short distance (some twitch), with a single press movement is smooth. 3. heating does not always start after the part is re-printed. (v3.2.1)

askmix, 2018-06-04:
mega 2560

askmix, 2018-06-04:
request developers, You can do to CURA showed current stance and tier in texts, if perhaps visually. ability to view the exchange of messages with the printer in text

ChrisTerBeke, 2018-06-04:
Please make separate issues for these different topics and and fill out the issue template for each of them. Like this we can't properly respond and track things.

closed 2018-06-04 by ChrisTerBeke

Ghostkeeper, 2018-07-13:
> 1. Сura 3.3.1 (ru language) do not start after pause, firmware marlin 1.1.8, win10 64bit. Сonstantly shows the temperature of the bed and hotend on 1 degree Celsius below the target. 2. With a quick [...]

I think this is more up to the printer than up to Cura. Cura just sends a command to the printer to heat up and wait until the temperature is reached. If the printer doesn't report that the temperature is reached, then Cura will keep waiting with sending more commands. For 2., Cura more or less immediately sends the command via USB upon pressing. It's up to the printer to execute it smoothly.

> 3. heating does not always start after the part is re-printed. (v3.2.1)

This one will be fixed for the upcoming release.