Ultimaker/Cura #3886: Request: Vary support material based on bed contact

AbeFM, 2018-06-04:
I love the ability to use soluble supports only for the interfaces - this can save a lot of material and add some reliability.

However, some parts have internal cavities which are better off using only PVA.

I would like to have supports *from the bed* made with extruder 1 (and soluble interface) but have supports which start on the model be entirely from extruder 2/

**Application Version**
Future

**Platform**
NA

**Printer**
Dual Extrusion

**Expected results**

[image]

ChrisTerBeke, 2018-06-04:
Would be a nice addition indeed :)

Ghostkeeper, 2018-07-13:
This introduces a circular dependency though.

Typically, PVA will have a 3mm or so horizontal offset. This means that a piece of support could be resting completely on the model for PLA, and thus get turned into PVA, but then get a horizontal offset that causes part of the support miss the model and rest on the build plate, so it must be printed with PLA, which means that it doesn't get the horizontal offset any more, and then rests only on the model again.

I don't see how we can have an elegant solution for this.

jackha, 2018-07-16:
The best thing I can come up with is to only allow this option when the horizontal expansion of both materials are set to the same value.

AbeFM, 2018-07-16:
That sounds reasonable to me.You would want them to stack nicely anyway.

You have to inflate what's underneath to support it anyway, if you don't.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2018-07-25:
Turns out you can already do this: Enable Support Interface and set the Support Floor Thickness to 9999mm.

Does that cover your use case enough? I don't think it's a large enough use case to warrant a setting and separate implementation to do this.

AbeFM, 2018-07-25:
That would technically make the supports out of the right material, but
then you're filling the part with super high glue percentage, as the
interface is typically very dense.

The inflation thing is reasonable, it's something that needs to get dealt
with.

In reality, the surface support typically comes off ok from the outside, so
I end up ripping that off, but it's hard on the pierce sometimes to do so.

Not the highest priority improvement, but something I think folks would
like.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2018-08-08:
Thought a bit more:
The inflated support could just change into PVA at the layer that it's only
there because it's supporting something above it, not because it's inflated
from something below. I don't know if that's how it works, but it seems
like if the bit next to you is PVA then you should be too, in that context.

If you're using PVA interface, then it seems easy, you're just printing PLA
from the raft (also PLA in this case), interfaces like normal. When you hit
an interface, everything's PVA, including the horizontal expansion.
Anything on PVA stays PVA, which fixes things if the lower interface is PVA.

Again, I don't know how it works, but it seems like maybe you only need to
keep track of if you were on top of PVA.

AbeFM, 2019-04-05:
Suddenly, I had a thought:

A) 3 Step process
    1) Slice the model with supports set to "Bed Only", generate in PLA
    2) Slice the model with supports set to "Everywhere"
    3) Subtract the support structure from 1), and do what's remaining in PVA

This might be technically challenging, but if I used the same settings between the two extruders (a big assumption), it seems like you could do it manually in a text editor if you tried hard enough, so it should be possible to automate.

B) The other method is more simplistic - if the thing you're printing a particular line of support on is the model, print with PVA. If the thing you're printing on is support, print in whatever the layer below you is.

Even if you expanded onto the model, you'd follow that rule to make the determination. I don't think it's as clean as "A)", but perhaps it's more easy?

Every time I mention this on a forum, there's a lot of interest. The volume of PVA that could be saved making something like a lattice sphere is significant, perhaps half of the supports are "outside" of the object and could be easily removed. I could certainly live with some restrictions on the support options, and I could stop using half the support I need just trying to save material.

Running some quick slicing to get numbers for stuff - I'm slicing this with relatively low 30% support density with 1 gradual support step, supporting at a 68* angle.

[image]

[image]

[image]

That's a 35% savings in support material, with a roughly 4:1 cost ratio.  Even with my super cheap material, it's a 25% overall print cost savings (not just PVA costs).

At 55 degrees support angle, other settings conservative as above, the costs savings grow, 168g PVA vs 278g PVA - that's a 40% PVA savings, And a 41% cost savings overall. It may well print faster, as well, making this really a factor of 2 in usefulness.

Sorry for the long post, I haven't put these thoughts out explicitly yet. I hope you find them motivating, or at least worthy of more discussion. :-)

Ghostkeeper, 2019-04-11:
> That would technically make the supports out of the right material, but then you're filling the part with super high glue percentage, as the interface is typically very dense.

You can set the support floor density to 10% and it won't be very high density, just the top part that actually uses the support roof.

> A) 3 Step process
> 1) Slice the model with supports set to "Bed Only", generate in PLA
> 2) Slice the model with supports set to "Everywhere"
> 3) Subtract the support structure from 1), and do what's remaining in PVA

Which settings should it slice for during step 1 and 2 though? The PLA extruder or the PVA extruder? And how would the user indicate that he wants to use those settings/that extruder? If the build-plate-only extruder has a higher horizontal offset than the everywhere-extruder, which extruder will be used for the overlap?

I think the support floor option is a better solution than this suggestion, also w.r.t. slicing time.

> B) The other method is more simplistic - if the thing you're printing a particular line of support on is the model, print with PVA. If the thing you're printing on is support, print in whatever the layer below you is.

This is basically what we do with support printing on build plate only. We remove areas of support above the print after generating all support. Problem is that the function of Support Horizontal Expansion is defeated by this, because you'll end up with tiny pillars of PVA support that won't have space to start extruding well and will topple over, UNLESS you perform the Horizontal Expansion after subtracting the areas but that then has the circular dependency that you'll be printing your build-plate-only material on top of the model.

AbeFM, 2019-04-25:
Could you do the expansion first, then the subtract?

You could also require the expansion of the upper layers be less-than-or-equal-to that of the lower layers, then the problem wouldn't exist?

I couldn't complain about 0.5mm of wasted PLA to save 17 cubic yards of PVA supports under the model.

I may not be following what you';re saying, but it seems like that would solve that specific issue?

Ghostkeeper, 2019-04-29:
> Could you do the expansion first, then the subtract?

That's what I thought you meant. It has the problems of:
* Which extruder's settings will it use?
* It'll create very thin slivers of PVA which will never extrude properly because they don't have horizontal expansion.
* Does the build-plate-only extruder also apply horizontal expansion towards the inside?

Also answer me this please: What is the qualm against the solution of setting the support floor to be very thick and changing that to a different extruder?

AbeFM, 2019-05-05:
Can you post a picture of the support you would generate? I am unable to picture it. I've done a bunch of slicing, with thick floors/cielings of various settings, none of them are even close to what I want. Putting in a large floor value is weird, too - I set it to 1000 mm, it was highlighted orange, like an error, and slicing took ~8 times longer than normal.

When it was one, ALL the support minus one layer was in PLA, and the floor thickness setting reverted to 40 mm, somehow.

Also, using floors forces me to waste a bunch of material again - I need dense floor settings because the floor rarely sticks to the part, and having several inches of floor (I guess this setting must be equal to the overall part Z-dimension) means that all of it will be at 90-100% infill. That's a FAR cry from the minimal support I'm able to create by using methods closer to what I'm saying.
[image]

Can you show me an object floating above the bed, supported on PLA, with only PVA in "internal" cavities? A hollow half sphere would work. I'm clearly not understanding your suggestion on how to do the slicing and would like to see it.

Here I put two 'cubes' stacked, you can see the support "floor" surrounded by PLA... expansion?
[image]

[image]

Re: "which extruder settings do you use" is answered by "lock expansions together  if you have to to make this work". Really I'm not quite following your statement:

> Problem is that the function of Support Horizontal Expansion is defeated by this, because you'll end up with tiny pillars of PVA support that won't have space to start extruding well and will topple [...]

I feel if you make the expansion the same, you're ok. 

Sorry for beating up this topic so much - I'm sure there's good reasons for what you're saying, but I don't see them yet. It seems reasonably trivial to produce sliced model with one setting, then the other, and only use the stuff there when "support everywhere" is turned on with "infill extruder" set to 1, and 0 for the "buildplate only" stuff - and make no other change.

Ghostkeeper, 2019-05-06:
> Can you post a picture of the support you would generate? I am unable to picture it.

Here is a project file: [test.3mf.zip](https://github.com/Ultimaker/Cura/files/3147770/test.3mf.zip) (rename to test.3mf).
[image]

Slicing this didn't take significantly longer or shorter, but I imagine that printing all that at 90% infill will cause slicing to take longer (as well as printing).

> I need dense floor settings because the floor rarely sticks to the part

There is no way to configure different sections within the support floor. The problem is starting to grow beyond the configuration possibilities of Cura now.

> Can you show me an object floating above the bed, supported on PLA, with only PVA in "internal" cavities?

See the above 3mf file.

> Here I put two 'cubes' stacked, you can see the support "floor" surrounded by PLA... expansion?

Looks like it.

> I feel if you make the expansion the same, you're ok.

We can't enforce PVA and PLA to use the same expansion. That would increase printing time and material usage for all prints with PLA support.

> It seems reasonably trivial to produce sliced model with one setting, then the other, and only use the stuff there when "support everywhere" is turned on with "infill extruder" set to 1, and 0 for the "buildplate only" stuff - and make no other change.

Yeah, it's not because the two polygons have different shapes then, one of which must be expanded with the horizontal expansion which causes part of that polygon to be above the build plate instead of above the model or vice versa. Maybe that angle of explaining it makes it more clear to you, the whole business with expansion.

AbeFM, 2019-09-04:
I had dropped this for a while because supports weren't sticking AT ALL... but with SmartAvionics' fix, prints now come out, support floors contact the model. But I'm still having issues getting non-soluble supports out of internal cavities, and printing the lower supports in PLA literally cuts the cost in half for a print. Even the most generic of filaments has ~$25 in 'wasted' PVA supports beneath a model.

I still fail to understand why you can't have a *mode* where the two extruders use the same expansion (horizontal expansion doesn't support anything, anyway, so why would  you do that? Perhaps incorrectly, but I only use it to keep supports from falling over from being too thin), then you do the process outlined below (slice twice, use correct extruder for anything sitting on a particular material.)

The suggestion you made, it DOES work, I can eliminate PLA within the model. It takes a LONG (~5x as long? A small crop ~100mm on a side takes ~8 minutes on an i7 @ 0.3mm layers, Real model ~180 mm/side ~18 min) time to slice, the smaller I make the floor the faster it goes, so knowing what you want to get rid of before you do it helps.

But I'm losing all the awesome abilities of the floor and support in general - I can't use gradual steps, I can't have a dense floor without also filling the model with solid PVA. We could add an additional "floor for the floor" to gain back the settings, but that seems to be WAY over-complicating things.

So *IF* the expansion settings were the same (I could live with them locked when in this mode), wouldn't it be approachable? Given that you can half the print cost, isn't it at least interesting?

I get the feeling that "if it's support, and it's sitting on PVA, print it in PVA" would just work. I believe you could make no other changes if you set certain variables the same.

I appreciate the work-around, but I don't think it negates the need for this improvement. 

**Please explain how slicing with all support set to PVA (Everywhere), and slice again with Support Infill set to PLA (Touching Buildplate Only), and subtracting the two how it's not possible to combine them.**

I would really like to have high density support floor and roof for good adhesion, support infill with gradual steps, and all of those in PVA when the support exists above a proper support "floor", and PLA otherwise.

AbeFM, 2019-09-06:
Can we export support as an STL or something similar?

Ghostkeeper, 2019-09-06:
> (horizontal expansion doesn't support anything, anyway, so why would you do that? Perhaps incorrectly, but I only use it to keep supports from falling over from being too thin)

We know that PVA without horizontal expansion makes the print fail often since PVA (at least Ultimaker's brand) takes a while to get good flow going, which is necessary to prevent the support from being too weak. After a retraction, it essentially needs some purging before it becomes effective. For other profiles we've been using it to make sure that the support is always at least one line width wide (so half a line width expansion).

Having separate modes for different features makes sense for the developer that made the mode, but for a user who wasn't involved it fractures the setting space completely, which makes it very hard to determine when certain settings are available and when they are not. We've always done our best to prevent that, though for some features (e.g. wireframe and spiralize) it's not possible. From feedback we're getting it is confusing to users when settings disappear for no apparent reason. We're going to remove wireframe sometime in the (hopefully near) future due to disuse. Additionally, locking settings for multiple extruders to become the same leads to ambiguities (which of the two extruders does it need to use, and what if you intended to use the other extruder for most of the print?) as well as bad printing (e.g. horizontal expansion as explained before).

> **Please explain how slicing with all support set to PVA (Everywhere), and slice again with Support Infill set to PLA (Touching Buildplate Only), and subtracting the two how it's not possible to combine them.**

Which of the two do you subtract from which?

AbeFM, 2019-09-06:
Ok, well we're at least hitting on the core of the different views here:
I print PLA without horizontal expansion on a variety of printers with a variety of extruders and don't have issues with flow like you claim. I basically can't get my PVA to print badly, but then again I use the cheap stuff. :-P

More seriously, wouldn't a better way to address that problem be Unretract Extra Prime? Although you risk a blob it sounds like the way to go - even moreso if you use something like combing and all those other "retract X amount if Y" features that already exist within CURA.

What settings would have to be linked for this to work? If you're expanding 1mm, you're telling me I'm going to spend more on extra PLA (only below the model) than I will filling the entire interior of the model with PVA? The later costs tens of dollars and the prior costs tens of cents.

I am being quite literal in the costs here, I get prints that go from $22 in material to 45.

re: Subtraction Method:
Slice entire model with "Supports on Buildplate Only", with support infill set to "PLA".
Slice again with "Supports Everywhere", and support infill set to PVA.

Subtract the first slice from the second (let those 'from buildplate only' PLA supports replace the PVA supports in the same locations - much easier if settings are "linked" together).

 Print what remains. I'm SURE if I sliced something where this all happened in a plane that I could demonstrate the idea in Notepad. If that's not clear I can dig for the pictures where I did this in Paint to demonstrate some long time ago.

Let me know if you need more!

AbeFM, 2019-09-06:
As an unrelated side note:
[image]
I believe those have been linked for me for many months, certainly as long as I've used them.

AbeFM, 2019-09-11:
[image]

If the values really are the same, I'm less clear on why my suggestion wouldn't work.

Similarly, if I set a giant floor, doesn't that already know where is "above the model" and where isn't? I could use a 10 meter thick floor, and if support infill was to be generated, check if it would overlap with this virtual thick floor and use that to determine the material?

The settings are linked, so I don't see why one can't be arbitrarily substituted.

Ghostkeeper, 2019-09-13:
> if it is set to be identical, wouldn't everything cancel out?

No, we don't compute forces or anything there. If two polygons are adjacent and both are expanding with 0.2mm, they will overlap with a strip that is about 0.4mm wide.

> When it's exanded, it doesn't "belong" to anything? I.E. If it's expanded from on-the-model, print in PVA, if it's expanded from the floor, print PLA.

We can track what polygons are expanded from. When expanding two adjacent polygons, we basically have four choices:
* Let them merge (our current solution).
* Let them overlap (produces overextrusion).
* Subtract one from the other (is ambiguous).
* Subtract the other from the one (is ambiguous).

> I believe this would improve the user experience. Aside from that, if you can think of a way to generate the "from the floor only" volume for an arbitrary model - even in external software, I'd be VERY open. I could print that as a separate object, or with it's own settings, then print the rest of the model with standard settings. I've been thinking about a projection in SolidWorks, Fusion, MeshMixer, but have yet to figure something out.

I have an OpenSCAD script that can do that! :grinning: Here it is: https://github.com/Ghostkeeper/SlicerTestModels/blob/master/support/generate_support_model_only.scad I've made a modification for you that makes it generate only the support that rests on the model. To use it, change the string at the top to wherever your STL is. Sadly it includes 2 Minkowski sums so it can take a good half hour to compute unless your model is very simple. Also, due to a current bug in OpenSCAD you have to flush the cache (Design -> Flush Caches) before rendering with F6.

> Is it possible to do a check: **When you're printing Support Infill, check what extruder's material you are printing directly onto and print the infill with the same extruder.** Nothing would get printed "into" the model that wasn't PVA. The worst issue would be little stripes of PLA alongside PVA near the edges. Much like your second bullet.

Yes, that's essentially how we're detecting which parts of support rest on the build plate. But again, that happens before we're applying transformation of these polygons for horizontal expansion, conical support, etc. (and for good reason).

> (even if it goes "into" the model, it'll be removable, although I suppose a slanted cone could print and lock itself through a narrow hole?)

Yeah but I think that case is rare enough that we can ignore that, really.

> The thought is that you do all the "where" calculations first, then pick the extruder at the end.

I think *this* is the crux of the argument here where our ideas clash now. You're saying to split off the part that rests on the model at the end of the process of generating support areas. I'm saying to split it off at the start. The reason why we currently split it off at the start is because otherwise some features won't work for parts of the print, like conical support or horizontal expansion. For conical support it's admittedly not really a problem; having a negative angle there is meant to help with stability but if the other extruder is printing next to it that's fine. But for horizontal expansion it is a problem since it's necessary to allow the filament to get flowing.

AbeFM, 2019-09-13:
> But for horizontal expansion it is a problem since it's necessary to allow the filament to get flowing.

Requiring the extra expansion is only to support advanced, proprietary features like under-extrusion. My printers unfortunately don't have that particular advantage and just spit out the amount of filament requested.

For those with the underextrusion feature, can't you use extra-restart to fix that? I wouldn't think UM users would want poorly extruded areas even if they are 'required', and that solving the issue with extra priming would be cleaner?

> We can track what polygons are expanded from. When expanding two adjacent polygons, we basically have four choices:
> 
> Let them merge (our current solution).
> Let them overlap (produces overextrusion).
> Subtract one from the other (is ambiguous).
> Subtract the other from the one (is ambiguous).

This is ambiguous because you don't know which to subtract? I'd say use the same logic as for the Prime Tower - always replace with soluble, it's safer.
What do you do now when the join distance connects two supports of different types?

It seems the issue is really boiling down to horizontal expansion on two different extruders and not knowing how to deal with it?  To continue your thought: If we just remove horizontal expansion for this special mode, things would work? I would be *well* served by that, as would anyone not using Support Expansion to make up for a lack of results by tuning in other areas.

Find me an UM to play with and I'll get it printing without support expansion. :-) Oh! You're talking about printing without a prime tower, maybe?

Thanks once again!

(P.S. Trying your script, having windows/linux issues - first time using scad.)

Ghostkeeper, 2019-09-17:
> For those with the underextrusion feature, can't you use extra-restart to fix that? I wouldn't think UM users would want poorly extruded areas even if they are 'required', and that solving the issue with extra priming would be cleaner?

Yeah I'd like to see more experimentation with this. But PVA is a difficult material to work with. If you see it extrude it really goes all over the place at first, water vapour produces bubbles in the filament, that sort of stuff. Perhaps they thought that the prime blob needs to be more of a prime line. Or perhaps a blob would knock over the support.

> I'd say use the same logic as for the Prime Tower - always replace with soluble, it's safer.

For dual-colour (or PLA+ABS or something) it still needs to make some choice, and preferably the same choice every layer.

> It seems the issue is really boiling down to horizontal expansion on two different extruders and not knowing how to deal with it?

The issue is with *all operations on the support polygons* after they have been generated. Horizontal expansion is the simplest example I can give. All operations on the support polygons will currently cause the polygons to merge due to how polygon unions, differences, offsets, etc. work. This includes:
* Support Brim
* Support Top Distance
* Support Bottom Distance
* Support X/Y Distance
* Support Distance Priority
* Support Minimum X/Y Distance
* Support Stair Step Height
* Support Stair Step Maximum Width
* Support Join Distance
* Minimum Support Area
* Use Towers
* Tower Diameter
* Maximum Tower-Supported Diameter
* Tower Roof Angle
* Enable Conical Support
* Conical Support Angle
* Conical Support Minimum Width

Settings that influence support areas but are exempt are Support Overhang Angle, Tree Support and its branch width settings, and Support Placement itself. Those are applied before the Support Placement setting is applied.

Many of these could probably be moved to before the Support Placement is applied, but care must be taken. PLA may have very different supportive properties than ABS, so maybe the X/Y distance does need to be set differently then between PLA and ABS.

We're also resistive against breaking up the settings space by enabling/disabling settings that to a user would seem to be unrelated, just because the algorithm can't be made to work. That's a different issue though.

GregValiant, 2026-09-10:
I'm going to close this.

closed 2026-09-10 by GregValiant