Ultimaker/Cura #3887: "Avoid printed parts while traveling" avoiding non-existent parts

AbeFM, 2018-06-04:
I made a large bed-leveling object, which includes a large spiral over most of the bed.

**Application Version**
3.3.1

**Platform**
win10

**Printer**
Custom Dual Ext FDM

**Steps to Reproduce**
Slice a single-layer object (probably with multiple layer objects, untested)

**Actual Results**
Printer does long, spiral travels across open space avoiding parts which have yet to be printed.

**Expected results**
Printer avoids printed parts, but does not avoid  empty spaces

**Additional Information**
[image]
Here you can see the travels already performed avoiding the still-being-printed spiral.

[BedLevelQuick.curaproject.zip](https://github.com/Ultimaker/Cura/files/2070130/BedLevelQuick.curaproject.zip)

smartavionics, 2018-06-05:
There's a regression in 3.3.1 that causes some combed travel moves to "go the long way round" to it could possibly be that you are seeing.

Ghostkeeper, 2018-07-13:
The project file doesn't seem to load in the current Cura version any more :confused:

Ghostkeeper, 2018-07-13:
That is not the problem. It doesn't just the travel moves avoid all parts, regardless of whether or not they have been printed yet.

jackha, 2018-07-16:
We've discussed it in the team and we found it not important enough to "fix". One of the reasons that you may want this behavior is possible oozing over not yet printed objects - then the oozing ends up in the not yet printed object.

AbeFM, 2018-07-16:
Reasonable. First layers are generally wasteful. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes