Ultimaker/Cura #3888: Sliced model covering up through-hole

slikk66, 2018-06-04:
**Application Version**
3.3.1

**Platform**
Mac OS

**Printer**
Cr10s

**Steps to Reproduce**
slice this part: https://printabrick.org/bricks/55982

**Actual Results**
The center "cross" hole is covered up

**Expected results**
the center "cross" hole should be open all the way top to bottom

**Additional Information**
https://imgur.com/a/nygZS84
images of the sliced model

smartavionics, 2018-06-05:
Freecad thinks the model is broken:

[image]

coreyfaure, 2018-06-05:
Confirmed to be the model, not Cura in this situation. https://www.reddit.com/r/3Dprinting/comments/8ondye/cura_weird_slicing_issue_covers_up_main_hole_in/

smartavionics, 2018-06-05:
Hi, if you are happy it's not cura, please close the issue. Thanks.

slikk66, 2018-06-05:
Seems to be the model, not cura.. closing!

closed 2018-06-05 by slikk66