Ultimaker/Cura #3889: "tried printing out of printing area" in Cura 3.3.1 but not in Cura 3.2.1

g-gollin, 2018-06-04:
**Application Version**
Cura 3.3.1

**Platform**
Mac OS 10.13.4

**Printer**
Ultimaker 2+ 
STL file 3v3v1_clay_stamp_4.stl.zip
[3v3v1_clay_stamp_4.stl.zip](https://github.com/Ultimaker/Cura/files/2070290/3v3v1_clay_stamp_4.stl.zip)
 included.

**Steps to Reproduce**
Open Cura 3.3.1, slice the STL file, then try to print the produced gcode file on an Ultimaker 2+. This yields the error message “tried printing out of printing area” even when the object is only a few cm in size.

This time open Cura 3.2.1, slice the STL file, then print the gcode file without problem on the same Ultimaker 2+.

**Actual Results**
Error message appears on printer after giving it the gcode file made by Cura 3.3.1, and job does not print. The Cura 3.2.1 gcode file prints fine.

**Expected results**
I would expect both versions to print fine.

**Additional Information**

diegopradogesto, 2018-06-05:
That means that your model is to big... but the model that you sent is very small. Please send the project file (File > Save project...) instead of the STL so we can see exactly the settings you have enabled and the configuration.

g-gollin, 2018-06-08:
Hi Diego,

Any thoughts on this strange issue?

thanks—
George

George Gollin
Professor of Physics
437d Loomis Laboratory of Physics
University of Illinois at Urbana-Champaign
1110 West Green Street
Urbana, IL 61801
+1 217.333.4451

diegopradogesto, 2018-06-11:
Oh, sorry. I can't find the project file. What you attached in the post was the STL instead. Can you upload the right file again?

g-gollin, 2018-06-11:
[cura_gollin_3mf.zip](https://github.com/Ultimaker/Cura/files/2090860/cura_gollin_3mf.zip)

g-gollin, 2018-06-11:
I just uploaded them to the GitHub thread.

thanks—
George

George Gollin
Professor of Physics
437d Loomis Laboratory of Physics
University of Illinois at Urbana-Champaign
1110 West Green Street
Urbana, IL 61801
+1 217.333.4451

Ghostkeeper, 2018-07-20:
I see the problem in your 3MF project.

You need to go to the Manage Printers... -> Machine Settings and there disable the "Origin at center" setting. I think you enabled this accidentally.

That setting caused the middle of the build plate to become the 0,0 coordinate origin. Anything towards the front left of the printer then became negative coordinates. When the printer reads this it uses the front left of the printer as 0,0 position, so those negative coordinates were outside the build volume.

closed 2018-07-20 by Ghostkeeper

g-gollin, 2018-07-20:
Yep- that fixed it! Thanks!

George Gollin
Professor of Physics
437d Loomis Laboratory of Physics
University of Illinois at Urbana-Champaign
1110 West Green Street
Urbana, IL 61801
+1 217.333.4451