Ultimaker/Cura #3890: Combing mode all routes along outer wall when it doesn't need to

smartavionics, 2018-06-05:
**Application Version**

3.4.99

**Platform**

Linux

**Printer**

S5

**Steps to Reproduce**

Load project slice, print, be disappointed with resulting print quality, investigate issue...

**Actual Results**

What's happening is that Cura has routed travel along the outer wall of the part and the resulting dribble has caused a lot of visual problems.

**Expected results**

Cura should route on the inner walls/infill and not the outer wall when it can.

**Additional Information**

[image]

[image]
[UMS5_combo_mount_ela.curaproject.3mf.renamed.zip](https://github.com/Ultimaker/Cura/files/2072097/UMS5_combo_mount_ela.curaproject.3mf.renamed.zip)

smartavionics, 2018-06-05:
The reason it's doing this is because on that layer the model narrows down in a couple of places (near the ends of the "arms") and there's only enough room for the outermost wall line and so the combing boundary has been set to the outermost wall for the whole layer when it only needs to be set to the outermost wall for the narrow region. For the rest of the model, the combing boundary could be an inner wall. Interestingly, for the combed travel moves that are used  while printing the skin regions, it does use an inner wall and not the outer.?

smartavionics, 2018-06-05:
@jackha, you have worked on the combing boundary code recently, any ideas?

AbeFM, 2018-06-05:
I had some issues where I was blaming extruder cooling... But I could get a LOT of that missing/defective outer wall. If you guys are short on examples I could try to work some up, but otherwise will just watch and learn.

jackha, 2018-06-06:
This is how it works and it may or may not have caused this:
- Try to comb using inner walls only. Success -> finished, else...
- Try to comb using all walls. If a single part is very thin, it combs everything using all walls, and not only the thin parts. Success -> finished, else..
- Travel outside, then in again.

smartavionics, 2018-06-06:
One thing you could do is to not use the centre line of the outer wall as the combing boundary but use the inner edge, that would move the nozzle slightly away from the edge of the part.

jackha, 2018-06-06:
I was worried about the computational time. Now it does 1 extra try before going to travel. But yes, that would be better. Still don't know if it's actually the cause though... Have you tried different combing modes?

smartavionics, 2018-06-06:
No, because my "no skins" combing is quite different to master (see https://github.com/Ultimaker/CuraEngine/pull/781 for details).

smartavionics, 2018-06-06:
What I mean is that it's not useful to compare no skins combing mode to all mode.

smartavionics, 2018-06-06:
Hi Jack, I think I have a possible solution, working on it, will let you know what happens!

smartavionics, 2018-06-06:
I have a working solution, will continue to test and clean up and if all well, will submit PR.

[image]

diegopradogesto, 2018-06-13:
Hey, @smartavionics, do you have a picture of the printed part after your fix? It will help to get an extent of the improvement.

smartavionics, 2018-06-13:
The above photo shows a part printed on the S5 and the slicer I am using for that doesn't have the fix. I will recreate the issue on a different printer and get before/after images.

smartavionics, 2018-06-13:
Hi @diegopradogesto, here you go. I have printed a section of the model that showed the problem. This is using another printer and another filament compared to above but the end result is the same.

Layer view without PR, travel goes along middle of outer wall:

[image]

Layer view with PR, travel goes along middle of second wall:

[image]

With PR on left, without PR on right, you can see the dribble on the outer wall has made a lot of roughness not only on the layers immediately above/below the holes but also on the inside corners:

[image]

diegopradogesto, 2018-06-13:
Oh, nice! I will add the PR to be reviewed and tested following the normal process and I will do some test prints.
Thanks!

jackha, 2018-06-14:
Very nice work @smartavionics !

smartavionics, 2018-06-15:
> Can I trivially transfer my printer (and slicing) settings from windows to Ubuntu to test PR's there?

The PR has not been incorporated in any release yet so the only way you can test it is to build CuraEngine from source.

If you have a sample project that has the problem, please share it if you can and I will see if the problem is still there with this PR in place.

ctzsnooze, 2018-06-19:
@smartavionics thanks for your work on improving combing - I just got 3.4 beta (OK slow) and your retraction with combing made a huge improvement on my thin wall TPU prints.

There were some instances of dribbling type effects like in your photos above, but overall the results were massively better.  No need to use Z-retraction with its associated stringing problems, lovely smooth interior and exterior.  Fantastic!

In relation to the above, I'd have hoped that combing moves could be, in order of priority:
- over the middle of the infill, if present, or
- over the centre of innermost wall, as per this PR.

Thank you!

diegopradogesto, 2018-06-19:
@ctzsnooze this fix is not yet in 3.4 beta, it is still in review process and it will be incorporated probably in the next release (not 3.4). However during the 3.4 development stage we (also together with @smartavionics) were working in some improvements in the combing, so maybe is related with what you see in 3.4 beta. 
I appreciate that you share your thoughts.

ctzsnooze, 2018-06-19:
Actually in 3.4 the retraction during combing is in place, and works great!  That's what's making the difference right now with my TPU prints.  This change would make it even better  :-)

Thanks!

closed 2018-07-31 by Ghostkeeper