Ultimaker/Cura #3892: Machine Control Panel Request

LilBub, 2018-06-05:
Hi,

It would be very convenient if Cura had a built in Machine Control Panel, much akin to what Pronterface has, Octoprint's Terminal Tab, etc.

This way users can send commands to their printer for things such as PID Autotune, calibration, etc.  

It would be great to do all of these functions within the Cura application and not have to use other applications to do so.

ChrisTerBeke, 2018-06-05:
Given that Ultimaker doesn't really support USB printing on Ultimaker machines, and our printers have things like calibration on-board instead, I don't think Ultimaker will spend time on this. But community contributions to improve USB/legacy printer control are always welcome.

closed 2018-06-05 by ChrisTerBeke

LilBub, 2018-06-05:
That's unfortunate.  I'll share that with the 3D printing community