Ultimaker/Cura #3894: Minimum Support Contact Area OR Minimum Support Layers

AbeFM, 2018-06-05:
I've been playing with soluble supports, and the adhesion is somewhat lacking compared to single material prints. This is not a problem with most prints (something with a big, gradual surface on bottom). However, when printing a downward facing spike, the contact area can be on the order of square microns and not enough to hold the part.

You can ramp up support angle, but you end up generating a lot where you don't need it.
You can add inflation distance, but it doesn't really support any higher up.

I propose enforcing support for a minimum amount of contact area - so if the first layer isn't >1mm^2 (or some number), continue printing perimeters until "Min Layers" or "Min Area" is achieved.

**Application Version**
3.3.1

**Platform**
Win10

**Printer**
all (but esp. multimaterial)

**Steps to Reproduce**
Slice with supports enabled

**Actual Results**
I included two examples, from the same print. One is the side of a leaf, the other is the tip.
[image]

**Expected results**
I would like to see sharp parts end up more like this (same print as above)
[image]
You can see here the first three layers are quite encased.

**Additional Information**
NA

AbeFM, 2018-06-15:
Is this something where more informal would help? 

Or is there some way to accomplish this I don't know about already built in?

jackha, 2018-06-20:
How would the minimum support area then be implemented? There is currently a horizontal expansion option which is also used for tiny areas.

AbeFM, 2018-06-20:
I'm not sure mathematically what makes the most sense.

My feeling is if you don't satisfy the minimum contact area on a current layer, then  you perform the horizontal expansion by an amount which would achieve it (or a fixed width - 2 o 3 line widths?), and increase one layer. Recalculate.

Th issue with horizontal expansion is that it doesn't continue up the sides of the part - it's just making the towers wider.

I would love to play with some setting which print up to the defined angle, plus two layers x 2 widths.

Ghostkeeper, 2018-08-06:
Even if Cura would add more perimeters (or otherwise more area) around the support, it's still going to have to adhere to the X/Y and Z distances so the support is still going to keep its distance from your model.

I can see the sense in a functionality like this if:
* You need to have some amount of minimum extrusion in order to get the flow going with this material between retractions. It's not just specific for support then. We have this problem with Ultimaker's PVA material, which is why we've set Support Horizontal Expansion to 3mm for that material. That functions fine.
* The supports needs to be stronger in places where it would otherwise be thin. That doesn't seem to be your problem though.

For now I think you need to just adjust the X/Y and Z distances of your support. For soluble support I suggest setting the Z distance to 0 and the minimum X/Y distance very low (and let Z override X/Y). Normal X/Y distance can be as high as you'd use for normal support.

AbeFM, 2018-08-08:
Ah! I see.

I'm still in the process of learning how to use PVA effectively - I was
mostly happy with setting the X-Y distance to 0 - 0.2 mm (under have an
extrusion width) - and was originally happy with the surface finish. Now
I've been getting some artifacting doing so (I fear my nozzles haven't
remained level), I set the distance higher.

The thought had been that with a small X-Y distance, the surrounding loop
(printed before the PLA if possible) would help constrain the part.

Thinking more, I imagine you could just take the area of the inflation, and
extend it up 1-2 more layers beyond where it would normally start, perhaps
even allowing it to come in closer (kinda a similar z-overides-x-y
situation).

I still feel this helps marginal prints, or situations with small sharp
points/lines/islands. Another slicer I've used has a special speed just for
small islands which shows I'm not the only one having issues with small
islands in dual material situations. It's not Feature of The Year or
anything, but this is my take on it.

Ghostkeeper, 2018-10-10:
> Thinking more, I imagine you could just take the area of the inflation, and extend it up 1-2 more layers beyond where it would normally start, perhaps even allowing it to come in closer (kinda a similar z-overides-x-y situation).

So you mean to extend support upwards next to the model that you're supporting? I don't really see how that would help with the extrusion rate in the lower layers though.

AbeFM, 2018-10-10:
I'm not sure what you mean by "next model" - but much like adding supports you don't really need for overhang's sake can keep the model from tipping/rocking as it gets taller.

Setting the overhang angle lower (more support) and then letting z override x-y as you said does accomplish supporting the material - but often you don't want to support such an angle because you'll end up doubling the support material/time.

The idea is *for small islands only* you use those settings (lower support angle, closer distance) in their immediate neighborhood.

So: If your island is less than min_island_area then you draw a boundry around it (layer-height / shallow-angle wide, i.e. 45 degrees or something overkill) and use those settings there.

That will let you do what you describe, but only in small areas about the steep points. Small, steep islands and downward facing edges will get 2-4 layers of support, a good base to print on.

Another way to think about it - you never want to print on something that doesn't have some minimum amount of contact area below it.

Ghostkeeper, 2018-12-13:
What you describe already exists though. It's called Support Towers. It looks something like this:

[image]

Maybe you need to check "Use Towers"? Or maybe it's not triggering in the cases that you would expect. That could be a bug.

AbeFM, 2018-12-13:
Please refer to my earlier pictures. What you showed has the problem I'm talking about.

When you put the second PLA layer down, it's on a microscopic nub of nearly unsupported PLA. You have big torque on it by virtue of x-y distance being greater than Z-distance, and  you quickly knock it over.

I want to see a small bucket enclosing SMALL islands and sharp edges. That will keep the parts from being dragged away and will help support them from the sides.

The same thing is accomplished by setting the support angle much lower than you want (set for 45* or so when you NEED support at 65), but that's massive support. If you support small islands specifically you will really help "unprintable" objects come out.