Ultimaker/Cura #3895: dual extrusion going past home after material layer change

sofa-bear, 2018-06-05:
so using a custom printer in cura version 3.4    3.3.1   3.21   3.0.4 and 2.7 (they are all the ones I have tested)
after the material changes in the model the print head homes but too far in the x axis It prints fine until it is just one material then it homes too far causing the belt to skip and then the rest of the print being offset. old cura anything before 15.04.6 is fine but the prints are messy with that.

smartavionics, 2018-06-06:
There is a PR waiting to be merged into Cura that fixes a problem with erroneous travels to 0,0 when using multiple extruders (https://github.com/Ultimaker/CuraEngine/pull/776). Perhaps you are seeing that issue?

ChrisTerBeke, 2018-06-06:
This issue has been reported several times here and there is indeed a PR to fix this.

closed 2018-06-06 by ChrisTerBeke