Ultimaker/Cura #3896: [3.3.1] Cura does not add M106 commands to start fans for custom RepRap printers.

bass4aj, 2018-06-06:
**Application Version**

Seen on v3.3.1, unsure of when it started

**Platform**

Windows 10

**Printer**

Custom Reprap printer (M3D Promega)
[CFDMP_3DBenchy.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2074707/CFDMP_3DBenchy.curaproject.3mf.zip)

**Steps to Reproduce**

1. Import File to slice
2. Ensure Enable Print Cooling is checked.
3. Set values for Fan Speed, Regular Fan Speed, Maximum Fan Speed, Regular/Maximum Fan Speed Threshold.
4. Set Initial Fan Speed to 0.
5. Set Regular Fan Speed at Height to 0.15 (one layer height)
6. Slice the model

**Actual Results**

The model is sliced, and an M107 (fan off) command is at the start of the gcode, but M106 (fan on) is never called throughout the file. M107 appears again at the end of the gcode file.

**Expected results**

The model should be sliced, and starting at the second layer, there should be a line that says M106 P2 S(some fan speed value). This should change at any layer that takes longer than the value set for Regular/Maximum Fan Speed Threshold.

**Additional Information**

Sliced file gcode attached for convenience.
[CFDMP_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/2074710/CFDMP_3DBenchy.zip)

bass4aj, 2018-06-06:
I am unsure if this only occurs on Custom Reprap printers, and which settings actually affect whether or not it works.

smartavionics, 2018-06-06:
Here's a tiny portion of your sample gcode file:

````
G1 F3000 E-8
G1 F300 Z0.4
G0 F3600 X202.475 Y184.047 Z0.4
G0 X202.176 Y192.607
;TIME_ELAPSED:76.210716
;LAYER:1
M140 S60
M106 S127.5
G0 F5400 X202.176 Y192.607 Z0.7
;TYPE:WALL-INNER
G1 F300 Z0.45
G1 F3000 E8
````

As you can see, it contains an M106 (there's lots of them in the file!)

Actually, there's too many M106s in your file because it repeatedly sets the same fan speed due to a long time Cura bug (which has recently been fixed).

If your fan isn't actually spinning, perhaps it's one of those fans that doesn't like PWM control?

bass4aj, 2018-06-07:
What the heck! I must have accidentally swapped out my files! That being said, this wont work because its not specifying P2... ran the command directly to my printer and it wont run the fan unless I call out M106 P2 S.... Does Cura support mapping that command to a specific P value?

smartavionics, 2018-06-07:
No, sorry, Cura doesn't but you could do that use a post-processing script.

pilotltd, 2018-06-16:
Post processing doesn't appear to work.

smartavionics, 2018-06-16:
Well it works for me and I could easily replace M106 with M106 P2 (or whatever). Sorry, I know nothing about the plugins so I can't suggest why it's not working for you.

bass4aj, 2018-06-19:
I found out it was an issue with fan mapping in my firmware. I was able to
fix the issue. Thanks!

jackha, 2018-06-20:
Glad you figured it out! Closing.

closed 2018-06-20 by jackha

mantoles, 2019-10-21:
@bass4aj can you describe what you did to fix the fan problem.  I am having a similar issue.  Basically cura does not turn my fan on.  If I pause printing and send the command :  M106 manually it works.  If I try to send M106 S127.5 the command does not work. It seems to only work at M106 S255.  Not sure if this is the same problem you are describing in this post or another problem.