Ultimaker/Cura #3897: [3.4.0-beta] OpenSCAD integration: Invalid file when loading OpenSCAD file

stelgenhof, 2018-06-06:
**Application Version**
3.4.0-beta

**Platform**
macOS 10.13.5

**Printer**
Irrelevant

**Steps to Reproduce**
0. Install the OpenSCAD integration plugin (via Toolbox)
1. From the "File" menu, select "Open File(s)..."
2. Select any OpenSCAD file (with the extension .scad)

**Actual Results**
Error message "Invalid File: Failed to load <NAME OF FILE.SCAD>"

**Expected results**
With the OpenSCAD integration I was kind of hoping to be able to load an OpenSCAD file :) However I may be doing something incorrect, but couldn't find any instructions how this plugin need to be used.

Some documentation would be great :)

Vandrasc, 2018-06-06:
Hi @stelgenhof, where is your openSCAD application located? Is it in the default installation folder? I experienced the same after I change the location of my application. Moving the openSCAD back to the default installation folder can do the trick. Let us know. 
Thank you!

stelgenhof, 2018-06-06:
@Vandrasc My openSCAD is not in the default installation folder. Let me try moving the application back to its default installation folder and see if it then works.

fieldOfView, 2018-06-06:
Pinging @thopiekar who wrote the plugin

thopiekar, 2018-06-06:
Sadly I can't give any macOS support at all. That's simply because I don't own a Mac.
The only chance how to get the plugin working is to add the OpenSCAD binary's location to $PATH. Windows enjoys the little extra that the plugin will look up in the registry which application is assigned to the file. Apples default application framework provides such a functionality, too, but this would need another extra Python module in Cura built-in.

In short: I leak in hardware and software at this point.
With a lot of effort I could bring it up, but I'm too busy at the moment for this.

Excuse the bad news on this, but that how it currently is 🙁

Anyway, regards from Hilversum, Netherlands ✌️

ChrisTerBeke, 2018-06-06:
Closing here as it's a plugin issue, not a Cura issue.

closed 2018-06-06 by ChrisTerBeke

Ghostkeeper, 2018-07-18:
See https://github.com/thopiekar/CuraOpenSCADPlugin for the source code (and issue reporting).