Ultimaker/Cura #3898: Octoprint plugin get

dimitribizakis, 2018-06-06:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

dimitribizakis, 2018-06-06:
Octoprint plugin always has the update button, i update and restart.
it's works fine but when i close and reopen Cura i need to do the procedure all over again.
Tried to manually install but the same thing happens.

fieldOfView, 2018-06-06:
First, please use the issue template and don't put the body text of your issue in the title.

There's a similar report here:
https://github.com/fieldOfView/OctoPrintPlugin/issues/80

Could you go to Help -> Show configuration folder, and open cura.cfg in a text editor?
Check if there's a line that says ```disabled_plugins = ,OctoPrintPlugin```. If there is such a line, close Cura (important!), remove that line and save cura.cfg. Next restart Cura.

closed 2018-06-06 by dimitribizakis