Ultimaker/Cura #3900: [3.3.1] Invisible font in the dropdown menus

jfrankenau, 2018-06-06:
**Application Version**

3.3.1 but also on 3.2

**Platform**
NixOS/Linux

**Steps to Reproduce**

Open one of the dropdown menus in the Print Setup sidebar.

**Actual Results**

Invisible font, see screenshot below.

**Expected results**

All availabe options in a readable font.

**Additional Information**
[image]

fieldOfView, 2018-06-06:
Do you have a dark system theme?

Vandrasc, 2018-06-06:
Do you also experience the same in 3.4? We just had a 3.4 Beta release. Thank you!

jfrankenau, 2018-06-06:
No, I am not using a dark system theme. However, I am using plain xmonad without a desktop environment and have just been testing to manually set `QT_QPA_PLATFORMTHEME` to `qt5ct` and it seems to work. I have no idea why other Qt applications are not affected by this. So it seems to more like a NixOS/xmonad issue.

I will test 3.4 later this day.

jfrankenau, 2018-06-07:
The issue persists with 3.4. I've also tested Cura's Ultimaker Dark theme which doesn't help.

alekseisasin, 2018-07-04:
I hope in the next Cura version the pyqt version will be updated and this issue will be solved for you.

ianpaschal, 2018-08-01:
> So it seems to more like a NixOS/xmonad issue.

Sounds like it to me too. Unfortunately this is far too narrow of a use case for us to debug or test, and we won't be spending any time on it.

Sorry about that.

closed 2018-08-01 by ianpaschal