Ultimaker/Cura #3904: Multiply selected model accepts large multiples without question

synchysis, 2018-06-06:
**Application Version**
3.3.1

**Platform**
Win10 (all?)

**Printer**
UM2+ (all?)

**Steps to Reproduce**
Select model and hit Multiply Selected Model. Try to enter 8. Enter 89 by mistake, then hit Enter.

**Actual Results**
User has to wait a long time for Cura to finish. Operation appears to take exponentially longer with larger multiples of parts.

**Expected results**
Cura checks if raw multiple entered is large (maybe >49), maybe as well as whether the multiple makes sense given the size of the model and the build area. Cura asks if user really wants to generate so many multiples of the model, allowing user to Proceed or Cancel the operation.

**Additional Information**
A simple formula for the maximum reasonable multiple could just be ((buildArea_xsize)×(buildArea_ysize)) / ((model_xsize)×(model_ysize)). Any number larger than that, or just generally a large number, could trigger the "are you sure" popup and save some time and headache for people using Cura to produce batches of parts.

diegopradogesto, 2018-06-29:
We discussed internally and we are not doing it for now. I will keep it open in case somebody want to work on it.

ianpaschal, 2018-08-01:
I don't think this is something we need to clutter our codebase with. At some point we can say that Cura requires its users to keep their brain switched on.

Also, this number would change a lot with different people's systems and there's no clear point at which it's "too much waiting." 2 seconds? 20 seconds? 200 seconds?

For things we feel really should be in Cura, but we don't have the necessary time for, we should use deferred but in this case no one thought it was worth doing hence the "won't fix", and so in that mindset, I'm going to close this. Can be reopened in the future if interest or rationale increase.

closed 2018-08-01 by ianpaschal