Ultimaker/Cura #3905: [master] Crashes on material creation

G-Pereira, 2018-06-06:
https://pastebin.com/EKFADEyk

Seems to be a issue with Qt.
I start on a fresh install I follow the initial setup to configure a new printer and then the material list is empty and crashed when I try to create one

Appesteijn, 2018-06-07:
Are you running from source? What OS?

LipuFei, 2018-06-07:
Hi @G-Pereira , looks like you are using Qt 5.12? We haven't tries anything above Qt 5.10.1 yet. The latest Qt + PyQt combination we know that works is Qt 5.10.1 + PyQt 5.10.

Perlovka, 2018-06-12:
Seems like this is not Qt problem.
When fdm_materials are not installed/found, cura will fail upon creating material with the following error:
```
2018-06-12 14:48:45,401 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-06-12 14:48:45,402 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-06-12 14:48:45,402 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/usr/lib64/python3.5/site-packages/cura/Machines/MaterialManager.py", line 528, in createMaterial
2018-06-12 14:48:45,402 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self.duplicateMaterial(material_group.root_material_node,
2018-06-12 14:48:45,402 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'NoneType' object has no attribute 'root_material_node'
```

Looks like it tries to duplicate nonexistent material and crashes (obviously). There should be some check before this.

clytle374, 2018-06-22:
I have had the same problem from source for a month.  Never got the materials to show up either.  From source on Gentoo linux.  Will try and help if someone has questions. 

Cory

Perlovka, 2018-06-24:
@clytle374 cura-3.3.1 been merged into gentoo tree some time ago, including materials

clytle374, 2018-06-24:
About a week after I finally got it built from source. lol, figures. Not sure why the materials don't show up, but I'll post another thread if I don't figure it out.

Ghostkeeper, 2018-08-07:
You may need to clone this repository into the `resources` directory and rename it to `materials`: https://github.com/Ultimaker/fdm_materials.git

Or better yet, clone it somewhere else and symlink it to the `resources` directory.
That should get the materials to show up again.

nallath, 2019-04-04:
Just tried to reproduce this and it no longer happens.

closed 2019-04-04 by nallath